portapack-mayhem/mayhem-firmware #1599: Accessibility fixes

alper061, 2023-11-21:
**Is your feature request related to a problem? Please describe.**
As a blind user, I find the current interface of the PortaPack Mayhem firmware challenging to navigate. The lack of accessibility features for visually impaired users greatly limits its usability for individuals like myself.

**Describe the solution you'd like**
I propose the implementation of a speak output feature for auditory feedback of menu options and functions. Additionally, the interface should be navigable using only buttons next to the touch screen.
This would greatly enhance the accessibility for visually impaired users.

**Describe alternatives you've considered**
The easy quick accessibility fix can be: include a quick-access feature or shortcut to switch the device into HackRF mode, or a detailed instruction on how to achieve this without using a touchscreen (e.g., "turn on device, rotate encoder once to the left, then press button, etc").
But  built-in speak output and button-based navigation would provide a more integrated experience.

**Additional**
As a blind user, I am willing to assist in the development and testing of these features.

alper061, 2023-11-21:
I have a question and a suggestion regarding the "Accessibility fixes" for the PortaPack Mayhem firmware.

Firstly, I would like to know if there is a capability to play audio files, perhaps from an SD card? If so, what formats are supported? This information is crucial for implementing a 'speak output' feature.

If the system currently does not support audio playback, I would like to inquire about the steps that would need to be taken to enable this functionality. However, if this capability is already available, then a possible solution could be to generate and store the names of menu items in audio file format on an SD card. Subsequently, if the 'accessibility option' is enabled in the firmware, the system could automatically start playing these audio files, thus aiding visually impaired users in navigation.

I look forward to your response and am happy to assist in the development and testing if needed.

NotherNgineer, 2023-11-22:
Good suggestion.  Mayhem firmware currently does not support playback of audio files on the speaker, but this support has been proposed in issue #1156.  https://github.com/eried/portapack-mayhem/issues/1156

htotoo, 2023-12-01:
Not the best option, but with tts, we don't need to generate a tons of wav file.

This is an example of embedded TTS service:

https://github.com/Flipper-XFW/Xtreme-Apps/blob/e896285c57caf626f759241a6dd1b5758943305d/sam/stm32_sam.h

this or something similar could be a more generic implementation.

gullradriel, 2024-01-04:
With recent serial developments this can be finally achieved and maybe more easily than before:

- a text to speech on the website side
- 3 static strings being overwritten by:  the last action (button, touch) for string 1  / last focused object name for string 2 / text value of object if any

Then a simple command for usb serial, which would print "action:" \n "object:" \n "text:"
We may even create a special type to add to widgets, that if not null is the accessibility name of the object, else fallback on object display name

gullradriel, 2024-01-04:
Just adding ideas as we're speaking about it on discord.

@htotoo found this to get a map of all visible objects.

We think we can add to serial commands, one to have the map, one for last action / focused object.

Smells good for at least an accessibility support using usb serial and a browser !