portapack-mayhem/mayhem-firmware #1603: HackRF on and off button

hydrapsy, 2023-11-26:
----
Before creating this issue, **do the following**: 
* Read the Wiki on booting: https://github.com/eried/portapack-mayhem/wiki/Won't-boot
* Read: https://github.com/eried/portapack-havoc/wiki/Update-firmware
* Watch carefully: https://www.youtube.com/watch?v=_zx4ZvurgOs
* (if you are not in Windows) also check: https://www.youtube.com/watch?v=kjFB58Y1TAo

----

**Describe the issue**
A clear and concise description of what the issue you are facing is. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'

**My Hardware**
Please specify what PortaPack hardware version you are using. 
You can find the list of versions here: https://github.com/eried/portapack-mayhem/wiki/PortaPack-Versions

**Affected versions**
Please tell us what version you are running. 
If your device is still functional, try the latest nightly release before submitting this. 
You can find the latest nightly version here https://github.com/eried/portapack-mayhem/releases

**Were you able to update the firmware before?**
Things might be confusing the first time, please check the video available on the link above.

**Can you try the upgrade with a different PC/Portapack/HackRF?**
If is possible, swap hardware and try again. Also, try different USB cables, even if the one you are using works fine for other purposes.

**Additional**
If the issue is difficult to explain, additionally to the text please include images and videos.

Radiomix2000, 2023-11-26:
Hardware mod needed.
You can use powerbank PCB (AliExpress) with on/off button, that's the most simplest way to get what you need.
Mine just has one button. One short push - switch on. Two short pushes - switch off.

sizzlik, 2023-11-29:
Gotta love when people just post a topic plus nothing filled out in the form. "HackRf On and Off button" ..is that a question? A statement? Cant you find the on/off button? Did you find an on/off button? Do you wish there were an on/off button? What even is an on/off button?

@Radiomix2000 Yeh..i got the same clone (bet OP aswell)..i wish there would be a way to change it. I would rather like to use the encoder knob as menu select button and (guess thats OP request) have a software on/off function..or long press the knob for on/off and click is button click. But unless the guys who made it make their stuff open source it wont happen im afraid. Doubt that any of the dev gods here got the time to reverse engineer and find a sollution.

eried, 2023-11-29:
This is not related with the firmware

closed 2023-11-29 by eried