portapack-mayhem/mayhem-firmware #1614: ADS-B RX STILL NO recievement?

execccuter, 2023-11-30:
Hello,

does the ADS-B RX still not work? I just installed the latest nightly version.

Can you help me out?

NotherNgineer, 2023-12-01:
ADS-B RX "should" work in all nightly releases since n_231020 per comments in #1510.  If it still doesn't work for you, please describe the issue you're seeing, and exactly which nightly firmware version was tested.

Flying-Pit, 2023-12-02:
nightly 231201 the ADS-B works perfect. I can decode a lot of Planes around Berlin (200km)

bicurico, 2023-12-07:
Flashed mayhem_nightly_n_231201_FIRMWARE and got no reception, either.

Flying-Pit, 2023-12-07:
> > nightly 231201 the ADS-B works perfect. I can decode a lot of Planes around Berlin (200km)
> 
> Which version is your Portapack? h1 or h2?

Hello, i use H1 and H2 and with both i have Reception. 
i have to say that i use an external LNA. without it i have very bad reception, with an external LNA i can decode more than 10 airplanes. it also depends on what kind of antenna you use.

bicurico, 2023-12-07:
This is to reconfirm that I am receiving ADS-B with n_20231201 on Portapack h1.
The reception settings, using the standard antenna, are LNA:40 VGA:32 AMP:0.
Hope this helps others.

NotherNgineer, 2023-12-07:
BTW, if your reception gets worse when Amp=1, it's possible that the amplifier is blown on your HackRF board.  Some more info here:  https://github.com/greatscottgadgets/hackrf/issues/974

closed 2023-12-22 by gullradriel

darknebular, 2025-11-10:
Same problem here with ADBS and Portapack H1, there aren't any plane. With 1.6.0 fw was perfect.