portapack-mayhem/mayhem-firmware #1621: HackRF application suggestion, Mayhem Portapack ISM signal analyser and decoder.

Eclipse-AU, 2023-12-03:
As you may know, the flipper zero has a ton of amazing apps created by the community around it. There is one Sub-GHz app that stands out, [Protoview](https://github.com/antirez/protoview).

Protoview allows you see the high and low pulses of the signal, show the device information of the transmitter (i.e. manufacturer, protocol name, code transmitted, raw data etc.), and the modulation of the signal. Having an app like this on the portapack would be pretty useful as sometimes you often get a signal from a device you cant decode or recognise. 

Although this could be finicky or complex, the hackRF can be just plugged into a pc and an ISM 433 MHz (i.e. RTL_433) decoder can be used. An existing app, [RTL_433](https://github.com/merbanan/rtl_433), only works with an RTL-SDR, but there is a GNU Radio program which can do something like this with the hackRF. The only problem with that GNU Radio app is that GNU Radio seems difficult and too complex for newbies to understand.

In conclusion, one of the most common use cases of the hackRF is using its transmitting capabilities for ISM radio operated devices and what if we got a Mayhem Portapack application that combines the flipper app [Protoview](https://github.com/antirez/protoview) and the RTL-SDR app [RTL_433](https://github.com/merbanan/rtl_433) to make an ISM device decoder and analyser on the go!

htotoo, 2023-12-04:
In my roadmap to create something similar, but without the rtl433 part. Yet I can't do that part, but i plan to port the flipper protocols, and a visual like the Protoview.

Eclipse-AU, 2023-12-04:
> In my roadmap to create something similar, but without the rtl433 part. Yet I can't do that part, but i plan to port the flipper protocols, and a visual like the Protoview.

ah awesome, I've played around a lot with protoview, it's a pretty good tool for the flipper. An rtl-433/protoview app and a waterfall display like the one that is you see when you select a button to press in the remote app, would be a nice addition to the app as well. Also, there was a bounty for this idea wasn't there? main site for portapack bounties is gone it seems.