portapack-mayhem/mayhem-firmware #1634: Repeater function

Nhscan, 2023-12-10:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen. Remember that adding stuff is always possible, but time is a limited resource for everyone. Check the wiki for more information how to compile the firmware and try to explore modifying the code yourself.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional**
If the suggestion is difficult to explain, additionally to the text please include images and videos.

I searched the open request I did not see anybody else ask for a repeater if they did I apologize for double posting. 

Repeater function would be really useful slipper zero recently got a repeater function which works fairly well but I can see it working better on the hackrf. hack RF can make it a lot more customizable where it will repeat and you can set the amount of time before it repeats the signal. You can kind of do it with the new repeater function on The flipper zero but it's kind of like short long or normal be better if we could set it in seconds or milliseconds. I don't know what other flipper is there a firmware is in but the one I'm running is rogue master. 

I'm not a programmer or else I would consider trying to do something like this myself somewhere of the hardware person. I don't have a lot of money but I would definitely throw $25 into having a working repeater function/app on the hack RF. 

So kind of specifics. Repeater function of signals like the record clone does now. Have the ability to set the time before it retransmits the last signal that it recorded automatically. Obviously have a way to set the game and maybe the bandwidth as well..

LeeroysHub, 2023-12-17:
I saw a SubGHZ decoder app PR sitting there. If that gets merged, a Key Repeater should be a pretty easy extension to add to that.

gullradriel, 2023-12-22:
In progress in https://github.com/eried/portapack-mayhem/pull/1658

NotherNgineer, 2023-12-28:
The repeater functionality added in Recon in #1658 might suffice for some users.

Longer term, I propose that we create a separate Repeater app (of course HackRF will never have enough power to serve as a real Ham Radio repeater -- but perhaps with a good amplifier...)