portapack-mayhem/mayhem-firmware #1637: ERT_RX stores to the SD card but doesn't display anything on the screen

jibanes, 2023-12-10:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
The ERT_RX seems to work fine, I see some entries logged to the SD card, but I do not see anything being displayed on the screen while running the app.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to ERT_RX
2. wait
3. nothing shows on the screen, only 0 32 32 & "ID/TP/Consumpt/Cnt"
4. go to sd storage
5. observe that some ERT entries are stored in logs.

**Expected behavior**
Have ERT data showed to the screen AND logged to the SD card

**Affected versions**
1.8

jibanes, 2023-12-10:
Same behavior with n_231210

gullradriel, 2023-12-11:
"Observe that some ERT entries are stored in logs."
Please paste us the parts of the contents too

jibanes, 2023-12-11:
Log:
19800101051423 SCM+ A34DC657B2B4C863060C46E27E98/2D80E4CE1FD849290A003306A28A
19800102022008 SCM+ A0CE3FB3BF9E1D7C3A154E0FA41D/40024012008682884A65B2084C64
19800102022058 SCM E317A0E3BC716B67282/05646524C2411186481

Screenshots below.
[image]
[image]

Thank you!

NotherNgineer, 2023-12-13:
These log entries look like garbage data, so the ERT app filters them out.  Possibly the signal strength is too weak.  When the signal strength is good, in the log files the digits after the / should be a string of 0's ending in a 1.

jibanes, 2023-12-16:
I let it go for a few days, I captured 311 entries in the log, as shown below, are you telling me they're all *bad*?
`
[ERT.TXT](https://github.com/eried/portapack-mayhem/files/13693232/ERT.TXT)

NotherNgineer, 2023-12-16:
The log data looks like random noise to me.  The default frequency of 911.6MHz should be close enough considering the app's wide receive bandwidth, but with the latest nightly you can tweak the center frequency (try 915MHz for example).

The ERT app currently only support IDM, SCM, or SCM+ data format.  The data from this meter might possibly be in Zigbee Smart Energy Profile format which I believe are encrypted and not currently supported by the app.

TBH, I've never seen the ERT app work with an electric meter (I do know it works with iTron gas meters).  You might try capturing some transmissions from your electric meter using the Capture app to analyze.

jibanes, 2023-12-16:
Thank you Mark, what application do you recommend to analyze the captures for ERT meters like mine?

NotherNgineer, 2023-12-17:
> Thank you Mark, what application do you recommend to analyze the captures for ERT meters like mine? Linux only.

If you find a good one, please let me know.  Personally I have only used Audacity to view capture files, as described on this Wiki page:  https://github.com/eried/portapack-mayhem/wiki/C16-format

jibanes, 2023-12-29:
Okay, so to summarize what's happening here:
I used rtlamr (on linux x86_64), center frequency=915Mhz, I get SCM+ data (approximately 20-40 rows per day of capture), checksum match; all data can be decoded and whatever meter/counter information is there, example entries look like:

```
{Time:2023-12-24T19:16:51.357 SCM+:{ProtocolID:0x1E EndpointType:0xAB EndpointID:  75502502 Consumption:        76 Tamper:0x4900 PacketCRC:0xF990}}
{Time:2023-12-24T19:40:51.866 SCM+:{ProtocolID:0x1E EndpointType:0xAB EndpointID:  75502502 Consumption:        76 Tamper:0x4900 PacketCRC:0xF990}}
{Time:2023-12-24T20:09:13.946 SCM:{ID:25098139 Type: 7 Tamper:{Phy:02 Enc:01} Consumption: 2407956 CRC:0x6F81}}
{Time:2023-12-24T20:34:19.902 SCM:{ID:25098139 Type: 7 Tamper:{Phy:02 Enc:01} Consumption: 2407980 CRC:0x4D64}}
{Time:2023-12-24T20:56:42.913 SCM+:{ProtocolID:0x1E EndpointType:0xAB EndpointID:  75502540 Consumption:       751 Tamper:0x4900 PacketCRC:0xB060}}
{Time:2023-12-24T20:56:51.402 SCM+:{ProtocolID:0x1E EndpointType:0xAB EndpointID:  75502502 Consumption:        76 Tamper:0x4900 PacketCRC:0xF990}}
{Time:2023-12-24T20:59:23.733 SCM:{ID:25098139 Type: 7 Tamper:{Phy:02 Enc:01} Consumption: 2408019 CRC:0x0E9C}}
{Time:2023-12-24T21:13:12.923 SCM+:{ProtocolID:0x1E EndpointType:0xAB EndpointID:  75502540 Consumption:       751 Tamper:0x4900 PacketCRC:0xB060}}
{Time:2023-12-24T21:18:26.852 SCM:{ID:25098139 Type: 7 Tamper:{Phy:02 Enc:01} Consumption: 2408043 CRC:0x2C79}}
```

On Mayhem's ERT, using the same antenna I used on rtlamr, same location, nothing goes to the display after a few days of capture; so Mayhem's ERT is probably not functional.

NotherNgineer, 2023-12-29:
It's nice to see that you do have some meters nearby transmitting in the Mayhem-supported SCM and SCM+ format, so here are a few more suggestions to try:

1.  To reset the ERT app to the default radio settings, try deleting the file SETTINGS\rx_ert.ini from the SD card before you run it the next time.

2.  Try increasing both the VNA & LNA gain values in the ERT app to 40, amp=0, and collapse the HackRF metal antenna fully (about 6.5 inches would be half-wave for 915MHz).  When I run the ERT app, I see many gas meters nearby with this setting.

3.  I don't know if there's a verbosity setting in rtlamr to dump the actual packet info.  If not, try the rtl_433 app with your RTL-SDR using the following parameters and capture the output for comparison if it detects any meters:
`  rtl_433 -f 915M -vv -R 149 -R 154 -R 160 -R 161`

jibanes, 2023-12-29:
Quick question, I see where VNA/LNA values are set, but where do you set amp=0 (or anything else)?

NotherNgineer, 2023-12-29:
The amp field was being covered up by the recently added frequency field.  I just pushed through a PR to fix that (will be in tonight's nightly release).

NotherNgineer, 2023-12-29:
> The amp field was being covered up by the recently added frequency field. I just pushed through a PR to fix that (will be in tonight's nightly release).

Actually the amp field visibility fix is in the new 1.9.1 release.

jibanes, 2023-12-29:
for the sake of completeness I use this antenna (which gave me the good results with rtlamr)
https://www.amazon.com/dp/B07R1XDQH8?ref=ppx_yo2ov_dt_b_product_details&th=1
note that rtlamr sets the center frequency to 912600155 by default.

jibanes, 2023-12-29:
I've been running on 1.9.1 for a few hours, no SCM/SCM+ so far; I've tried various VNA/LNA combinations and set the centerfrequency to the same as rtlamr (912.6)

NotherNgineer, 2024-01-04:
It would still be interesting to see the packet data displayed in verbose mode from the rtl_433 app as I mentioned above.
Also I should ask if other apps such as the ADS-B RX app are receiving and displaying a list.

Rf-guy-91, 2024-12-08:
Hello, I had the EXACT same issue and after weeks of trial and error I was able to get it working.
912.600MHz LNA and VGA set to 40.
In my neighborhood, nothing shows up on the ert rx application. It wasn't until I started driving around to different neighborhoods that it started working.
Another thing that helped me was the "search" app set from 902.000 to 928.000MHz. You can see what's popping up most frequently and try those frequencies in the ert rx application. 
I know this is over a old, so I don't know if anyone will see this comment. But it's still open so I figured I would try to help.
[image]

gullradriel, 2024-12-20:
Main answers for that issue are:
- use correct antenna
- use LNA and VGA at 40 (the app is a bit deaf)
- use correct frequency (identify them using Search app)

closed 2024-12-20 by gullradriel