portapack-mayhem/mayhem-firmware #1639: r9 devices, can not activate in current fw ,  the CLK out .

Brumi-2021, 2023-12-10:
Hi,  I realisded , that current  fw -example nightly 9th of December - 
* can activate CORRECTLY  the CLK OUT port  in all  previous r1 to r8  hardware platform that are using SI5153C 
1-) It can be activated through , Settings --> Radio 
[image]

2-) Or also possible to active through top menu icons (title bar) .
[image]
 When not activated through top icons menu ) ==> we got no signal through that CLK out port  (as expected)
[image]

(when we activate that icon )  ==> we got nice squared signal 3V3 pk-pk , through that CLK out port  (as expected)
[image]
 

* But this feature ,  it is not working yet in the r9  platform that is using  SI553A version . 
* (So far , confirmed in two r9 units , Bernd unit , and mine) 
[image]

We guess that it should be just fw register settings issue ,  SI5153A  in  r9 platform. 

Cheers,

NotherNgineer, 2023-12-22:
Not sure why yet, but a Si5153A register dump indicates that firmware is trying to enable CLK OUT on the CLK3 pin (which does not even exist on the SI5153A) instead of on the CLK2 pin (which is the output we _need_ to enable on the R9 boards).  Maybe this part of firmware doesn't think it's an R9.  BUT, these registers also seem to indicate that the CLK2 output is _already_ (always) enabled.....  I'll keep debugging.

[image]

NotherNgineer, 2023-12-22:
So I found that ClockManager::enable_clock_output() was messing with CLK3 on all boards due to a missing check for R9.  But, after changing this function to control CLK2 for R9, I find that the CPU immediately hangs whenever I disable CLK2 output.  Still debugging, just FYI.

gullradriel, 2023-12-22:
It's still progress finding the bugs !

Brumi-2021, 2023-12-22:
Hi @NotherNgineer , i have been enjoying about that issue for one week reaching similar conclusions .
Later I will contact you to try to exchange my results and accelerate its solution . 
But the good news is that the hw r9 is ok , the problem is purely sw .

NotherNgineer, 2023-12-24:
Status update:  I'm finally set up to view the HackRF v9 schematics and successfully added code to control the GPIO bit to enable CLKOUT on r9 boards.  I also see that CLK2 is also routed to P4_7 for the purposes of measuring the CLKIN frequency.  I've assigned the issue to myself.

NotherNgineer, 2023-12-24:
After further investigation, I've found that the GP_CLKIN, also known as MCU_CLK, is used as a reference for many other clocks in the CPU, including PLL0_AUDIO_CTRL, PLL1_CTRL.  IDIVB/C/D depend on those clocks, and then BASE_PERIPH_CLOCK, BASE_M4_CLOCK, BASE_AUDIO_CLOCK and others are based on IDIVx clocks.  BASE_M4_CLOCK is then used for all sorts of things.

Changing the frequency, disabling the MCU_CLK, or turning off CLK2 output at the Si5351A on r9 boards would require some LP43xx clock reconfiguration to use another clock reference for the above (such as the LP43xx's clock crystal).

For r9, firmware can simply leave the Si5351A's CLK2 output enabled permanently and turn CLKOUT on & off using GPIO(0,9) corresponding to pin P1_2, the CLKOUT_ENABLE pin.

BUT changing the CLKOUT frequency on r9 boards will be problematic without removing the existing dependencies on GP_CLKIN mentioned above.

SO the question is whether it's OK to lock the CLKOUT frequency to a fixed setting of 10MHz on r9 boards (simplest firmware change), or shall I work on reconfiguring the LP43xx to eliminate this dependency on GP_CLKIN so that users will be able to adjust the CLKOUT frequency to other values?

My recommendation would be to limit the CLKOUT frequency to 10MHz for now, at least as the simplest interim solution.

Brumi-2021, 2023-12-24:
Based on the merged  #1671 ,  we can close that issue .
Thanks @NotherNgineer , that investigation , had many hidden  misteries and hypothesis  , but we could make clear one 
by one ...  but original  hackrf submodule code, and all hackrf debug tools , were the key to fix it ! 

Not sure if I will ever use that CLK out in r9 or not,  
but once we found that it did not work , it was a good challenge to fix it , and   we did not stopped its investigation till solved it !   So @NotherNgineer , really good job !!!!  thanks ,  you solved the mystery ! 
In next days I will add some oscilloscope pictures , just to record it , (but just to see the squared signal quality in several freq.)
Merry Christmas !

closed 2023-12-24 by Brumi-2021