portapack-mayhem/mayhem-firmware #1645: Radio settings frequency correction

Nhscan, 2023-12-12:
12/12/1023 build if you try and to radius settings frequency correction if you try to make it positive it will not retain the positive settings it will only retain negative. So if you switch it to + then hit return. It will not retain the settings on TX mode settings. This is from the last nightly build. 

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
A clear and concise description of what the bug is. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'
3. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:
* Latest Nightly release:
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

gullradriel, 2023-12-13:
Confirmed. There is something funky going on here.
I'll check and fix tonight, does not seem to be too much.

gullradriel, 2023-12-17:
https://github.com/eried/portapack-mayhem/pull/1649 is fixing it. 
it will be available in next nightly, or as soon as merged if you're building yourself.
Thanks for the report.

closed 2023-12-18 by NotherNgineer