portapack-mayhem/mayhem-firmware #1654: Can't get out of BLE RX

loewal, 2023-12-20:
**Describe the bug**
Unable to leave BLE-Rx

**To Reproduce**
Steps to reproduce the behavior:
1. Navigate to Receive ----> BLE-Rx
2. wait a while receiving
3. press simultaneously left+top to get out this section
4. unable to leave.....

**Expected behavior**
Go to Receive

**Affected versions**

* Latest Nightly release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

Brumi-2021, 2023-12-22:
Yes agree , it happens in the recent  nightlies and also in the 1.9.0

gullradriel, 2023-12-22:
I found out why, it's related to latest USB serial modifications.
It's working fine if you're having a serial terminal connected.

gullradriel, 2023-12-22:
Should be fixed in https://github.com/eried/portapack-mayhem/pull/1664

I added a checkbox to activate USB serial logging on demand only until we find how and why and fix the USB serial hang when nothing is attached.

Tested by me, but you can test using the pack here: https://discord.com/channels/719669764804444213/722101917135798312/1187685331823251496

iNetro, 2023-12-22:
> Should be fixed in https://github.com/eried/portapack-mayhem/pull/1664
>
> I added a checkbox to activate USB serial logging on demand only until we find how and why and fix the USB serial hang when [...]

Thanks @gullradriel I figured this was the case but I didn't have time to check.

gullradriel, 2023-12-22:
No problem :-)

closed 2023-12-22 by gullradriel