portapack-mayhem/mayhem-firmware #1657: flip / rotate screen

D0lby55, 2023-12-21:
hello!! i thank all of you for the fabulous work on this  firmware that you share! i had a suggestion for the screen is it possible to get an option to flip or reverse the screen to get a reading more comfortable on some apps?? or please if someone know another type of screen a bit bigger that be compatible to be adapted to the connector and firmware ???

sizzlik, 2023-12-27:
I asked in the past for that option. Would be nice to have an option to hold the PP sideways, but seems its pretty much hardcoded that way or would require extensive re-coding of all apps and waste of the limited space to implement that option. Now that apps can get exported to sd, size might not matter  that much (thats what she said) but still would require lots of work im afraid. Maybe someday..?

D0lby55, 2023-12-27:
not for all the apps but  you are right! that's certainly too much code for the device...
for the moment my glasses help me to get a better screen resolution :)

joeburn69, 2024-02-16:
I'm really interested in this idea as the case I'm building would be great if the main antenna port was upside down