portapack-mayhem/mayhem-firmware #1662: H1 with 1.9.0: ADSB-RX still not working, story continues....

voxo22, 2023-12-21:
I do not understand... latest working ADSB-RX was in 1.7.4....worked like a charm. From then on, simply useless. With just the same settings, same antenna - no aircraft detected, red level meter still almost at max - useless (even with no antenna connected), strange two blinking dots at the very right. What you guys did with it? What does it all mean? ADSB had no problems before 1.8.0, so why now still has? I live near airport, so I have rush hour almost every second.
Screenshots taken right now, immediately one by one (change firmware between).  

1.7.4
[image]
1.9.0
[image]
1.9.0
[image]
1.9.0
[image]

D0lby55, 2023-12-22:
Hello! i do not know if the rx_adsb.ini file be located into the "setting" folder in this 1.7.4 version but you must found it and erase it before launching the 1.9.0! old ini file can stuck the app.

Brumi-2021, 2023-12-22:
Hello @D0lby55 , that fixed your problem with Adsb rx I  1.9.0 ? I tried yesterday night it  and seemed not to fixed mine. Moreover in some older nov nightly , all is ok (no dropped samples) , but in that 1.9.0, yesterday night test -with low air traffic- no reception and I could also see with DFU-debug button , growing dropped  M4 samples and 99% M4 CPU . (In previous nov nightly around 88% M4 and no dropped samples)
I will recheck today .

voxo22, 2023-12-22:
Thanks @D0lby55 that did it! I have tested some nightly versions before, but 1.9.0 did not worked with previous .ini. I also simultaneously compared two HackRFs in ADS-B Rx mode, One with 1.5.3 and one with 1.9.0 and their results are now pretty same. Thank you.
@Brumi-2021 if you do not have any other problem with the app, you can close this issue.
Peaceful holidays to everyone.

D0lby55, 2023-12-22:
thank you! i had no pb with ADS-B RX ! but a similar pb on another app and i solved it by doing this.

Brumi-2021, 2023-12-22:
Hi everyone , 
It is true that deleting that adsb_rx.ini file in fw v 1.9.0  , solves also my reception ADSB rx  problem .

But , when debugging -pressing DFU_debug_button inside App) that 1.9.0 ADSB RX App ,(and latest nightlies) , I can see 
M4 usage of around 99% and  some missed M4 samples .   But when I use my local  compile binaries . (with higher gcc arm... v10,3 , ) I got a correct  M4 usage of 87% and no dropped samples in that ADSB Rx app . (strange , but it works better) .

Just in case that you want to try , I upload here , a fresh binary (that will appear in tomorrow's nightly ) 
* current  next version compiled locally with my gcc arm v 10.3  (with healthy 87% M4 usage inside ADSB Rx , in my set ) 
* and it has also another small counter measure, #1664 . solving the "exit hang up from BLE Rx app" , just released this morning , 
* 
[portapack-h1_h2-gcc10_3_next231222.zip](https://github.com/eried/portapack-mayhem/files/13751732/portapack-h1_h2-gcc10_3_next231222.zip)
[APPS.zip](https://github.com/eried/portapack-mayhem/files/13751734/APPS.zip)

Then if you agree, let me close by now this no ADSB RX reception

closed 2023-12-22 by Brumi-2021

NotherNgineer, 2023-12-22:
If anyone still has a copy of an old rx_adsb.ini file that doesn't work with 1.9.0, I'd be curious to see the file contents.