portapack-mayhem/mayhem-firmware #1665: Rx Search app not catching frequency on 1.9 release

Nhscan, 2023-12-22:
----
I kinda remember rx search working on before release it would catch the RX and display it fine down below with the info. Now no matter what it receives it will at most show it like it is trying but never drops it to the list below. I know the rx search app is still under development I just know it worked better for me a few nightly builds back. 

If this is a repeat question or already listed please delete sorry but I tried to find it.

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
A clear and concise description of what the bug is. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'
3. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:
* Latest Nightly release:
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

gullradriel, 2023-12-22:
I have catches when I try. 
Check your settings, and that external TCXO is set.
Sometimes after an update it's breaking the settings.

gullradriel, 2023-12-26:
How is it now ?
Still problems?
Have you tried using the nightly?

Nhscan, 2023-12-28:
Been a little busy with Christmas but definitely going to try it tonight if I get time thanks for the reply and hopefully it's fixed.

gullradriel, 2024-01-05:
Did you had time to check ?