portapack-mayhem/mayhem-firmware #1666: Does not boot without connection to power source

h0us3c4ll, 2023-12-22:
**Describe the issue**
does not boot if not connected to power

**To Reproduce**
Steps to reproduce the behavior:
try to boot the device, it does not. 

if i attach to power source (USB or mains line), it boots fine and i can use.  it is fully charged. 

fyi: if connected to power, i also cannot turn it off. but that is OK

**My Hardware**
HackRF Portapack H2 HackRF One

**Affected versions**
v1.8.0

gullradriel, 2023-12-22:
Your internal battery is dead/not charging

h0us3c4ll, 2023-12-22:
hi gullradriel, thank you. how can i fix it? it is working fine if i unplug power source **after** booting

gullradriel, 2023-12-22:
I would try to update, clean settings, test if PMem is good

h0us3c4ll, 2023-12-22:
how can i test that? also, but this is different issue, if i connect it to my PC, windows does not detect it.. so maybe indeed issue somewhere. thank you

h0us3c4ll, 2023-12-30:
Hi d0lby55, it works now, i did 5h full charge. maybe battery issue, thank's!

closed 2023-12-30 by h0us3c4ll