portapack-mayhem/mayhem-firmware #1673: transmitting issues

Amarthol78, 2023-12-25:
Bought a hackrf and a portapack h2 a couple weeks back, have not been able to transmit from the portapack. I get a signal but it has no content, I tried bluetooth spam to my phone but the phone never showed anything so then tried on the PC with the same effect. I then tried POCSAG transmitting to my flipper and it shows a signal but receives nothing. The other thing that then happens is when I try to transmit when I go back to an application to receive but I get nothing, I need to double click the wheel to turn it off then turn it back on to be able to receive again.
Now, playing with gnu radio companion on Pentoo I can transmit to 88.7 fm and pick it up on a radio, then switch back to receiving with no problem.

I have a hackrf r9 and a portapack h2 both from hackerwarehouse, I used firmware 1.8, 1.9 and now with the 12-25-2023 nightly build all to the same effect.

Any help would be appreciated!

gullradriel, 2023-12-25:
Update your CPLD (see the wiki)
Reset your settings (see Factory defaults in the wiki)
Be sure your coin battery is working or use pmem to SDCARD (on the wiki too)
Else it may be a faulty device

Amarthol78, 2023-12-25:
YES!!! THANKS!!! The CPLD reflash fixed it!!

gullradriel, 2023-12-25:
You're welcome

closed 2023-12-25 by gullradriel