portapack-mayhem/mayhem-firmware #1677: Frequency of TEDI/LCR

neverbetter34, 2023-12-26:
I read the section on the wiki about TEDI/LCR. However, I could not understand exactly how the tool works. 

The most important question in my mind is how to find the frequency. Is there a frequency scanning option for TEDI like in Looking Glass? I would be very grateful if someone who has used this feature of the device before could provide information on this subject.

D0lby55, 2023-12-26:
hello i can only help you with this links from WAYBACKMACHINE:
website: https://web.archive.org/web/20071012213437/http://equidyn.fr/
forum: https://web.archive.org/web/20071012214804/http://equidyn.fr/forum/  

i think that you could capture the frequency if you are near a road billboard in RX mode with the SEARCH app.

neverbetter34, 2023-12-26:
Thank you. I would be happy if there are others who would like to share their experiences on this subject

D0lby55, 2023-12-26:
est tu francais? il faut trouvé un panneau routier et ce posté a coté avant les heures de pointe au sortie de bureau, quand ca va bouchonné! ils changeront le message pour un "ralentissement" ou "bouchon" ou "accident" et la tu pourra avoir le signal!

le probleme c'est qu'il y a plusieurs signaux pour differents equippements...faudra les noté et testé un message.

neverbetter34, 2023-12-26:
I'm not French. But thanks for your help

QwartzNetwork, 2023-12-26:
> est tu francais? il faut trouvé un panneau routier et ce posté a coté avant les heures de pointe au sortie de bureau, quand ca va bouchonné! ils changeront le message pour un "ralentissement" ou "bouchon" ou "accident" et la tu pourra avoir le signal!
> 
> le probleme c'est qu'il y a plusieurs signaux pour differents equippements...faudra les noté et testé un message.
> 
> you have to find a road sign and wait before rush hour at the office exit, when it’s going to jam! They’ll change the message to "slow down" or "accident" and then you can get the signal! the problem is that there are several signals for different equippements...You have to note frequencys and test them with Tedi/Lcr.

C'est dommage qu'on est pas une fourchette de fréquences qui pourrait grandement faciliter les recherches

D0lby55, 2023-12-26:
equidyn = équipements dynamiques de la route
si l'on tape cela dans un moteur de recherche on a beaucoup plus de résultat, peut etre peut on y decouvrir plus d'info, je regarderai mieux cela plus tard...

closed 2024-04-08 by gullradriel