portapack-mayhem/mayhem-firmware #1679: White Screen H1R2 on 1.9.0

Chaminoux, 2023-12-26:
Portapack : H1R2 
When I boot after updating the firmware to version 1.9.0, a white screen is displayed, and it is impossible to boot.

That's what I tried : 
1. Update CPLD
2. SD card formated 
3. Files copied to SD card
4. Change of the coin battery
5. DFU with (nightly_231226)
6. Flash with the same nightly

- Try to boot with a fresh SD card  => white screen
- Try to boot with a fresh SD card with the boot key (all tested one by one) => white screen
- Try to boot without SD card with the boot key (all tested one by one) => white screen

The Portapack H1R2 works with the firmware 1.8.0 (and previous) using this boot key :
1. Hold left
2. Power on the Portapack
3. Release left after 5 seconds
4. Hold the reset button
5. Hold OK
6. Release OK after 5 seconds
-> The screen is black for 1 second, and the Portapack boots up

If I try the same process with a nightly or the 1.9.0, it doesn't work either.
All the content of the sd card is the same version of the nightly or release.
The "hardware" folder is present so the sd card is ok (the contente of the settings file is the number zero: 0 )
I have try to do that on windows 10 and linux (unbuntu).

D0lby55, 2023-12-27:
hello did you read this and test it ?? :

https://github.com/eried/portapack-mayhem/wiki/Won't-boot

Chaminoux, 2023-12-27:
Yes.

D0lby55, 2023-12-27:
i had this pb that i solved by doing this, this not worked the first time i tryed this method...or try to downgrade the firmware one more time before doing it.

AC9DJ, 2023-12-27:
Replace the button cell battery on the portapack board. I have run into this several times after the battery died.

Chaminoux, 2023-12-27:
as I already mentioned in the first post, I have already carried out this manipulation

NeyZeeN, 2023-12-27:
Hey guys, 
Same issue here with **H1R1**.
I opened it to take measurements for a battery and speakers and after reassembling it, this white screen appeared.

I double checked if I something was wrong..

Tried to downgrade and nothing changes.

Now it's stuck into HackRF mode (Thank God I can still use it on Windows).

Brumi-2021, 2023-12-27:
Hi @NeyZeeN , if you just removed coin cell battery to measure it and later reassembled , it sounds some corrupted issue about ram persistent memory . I think trying the top  steps of "won't boot " , you should be able to recover . Try it with or without SD card .  Later on , once recovered you will be able to decide in settings ->p memory mgt the use of sd card or persistent memory ram or reset them.

NeyZeeN, 2023-12-28:
Hello @Brumi-2021, 
Thank you for your answer. I just tried and still the same issue..
The battery cell button is charged with ~3.2V.

jLynx, 2023-12-28:
So just confirming. If you haven't yet, please try these steps.

Remove coincell and remove SD card. Now try boot with all available button boot commands without the coincell and SD card and see if you get any progress from there. Please let me know the outcome.

NeyZeeN, 2023-12-28:
Hello @jLynx, 
Just tried your method and I still have the same Issue.

Brumi-2021, 2023-12-28:
Hi @NeyZeeN , just to keep discarting different possible root causes , I will comment two more test : 
1-) (the most probable case )  It could be that the flashed fw had some communication problem when writing and therefore, maybe the SPI flash binary is corrupted. 
==>  put your unit in DFU and try to rewrite again any latest Mayhem fw .

2) (rare case ) if the previous option does not boot correctly , you can try another idea . Switch off the unit , keep doing "won't boot" steps but this time do not power up from its  battery.  Power it , just connecting it to a  charged usb power bank or pc laptop.  (In some units ,  when we have low internal Lipo battery voltage , the PP TCXO clock might be  not well detected (in some units)  and can produce hang up . But if you power up the unit from strong 5V USB, it might be detected correctly) . If that is your tolerance voltage  case -and you could boot correctly with the help of  external usb battery- I will recoment to go to settings --> Radio and disable always ext TCXO.(to avoid that problem when your battery voltage is slightly low )  

(1) is the most probable case of your problem ,
But just in case, if (1)  does not work you can also try (2) . 
Good luck !

NeyZeeN, 2023-12-28:
Hello @Brumi-2021, 

1) The DFU method returns me an error : 
**unable to read DFU status after completion
can't detach
Resetting USB to switch back to runtime mode**

Here is the full terminal : 

*** Run HackRF firmware in RAM via LPC DFU ***

This is used to "unbrick" your HackRF, if you are no longer able to use
HackRF tools to flash or operate your HackRF.

Connect your HackRF One to a USB port on your computer.

Hold down both the DFU and RESET buttons on the HackRF.
Then release the RESET button (closest to the edge).
Then release the DFU button.

Press any key to continue . . .

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

DFU suffix CRC does not match
A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download        [=========================] 100%        42564 bytes
Download done.
unable to read DFU status after completion
can't detach
Resetting USB to switch back to runtime mode

Press any key to continue . . .

2) I do not have a battery. Therefore, my main computer has 4.90V and my laptop outputs 5.10V. Still no difference between them..

Brumi-2021, 2023-12-28:
Hi. Again. Looking this sentence

"DFU suffix CRC does not match"
I guess you need to follow Hackrf instructions,

https://github.com/greatscottgadgets/hackrf/issues/380

And once recovered , you will need to reflash with Mayhem fw

NeyZeeN, 2023-12-28:
Hello @Brumi-2021,

I did what I understood as flashing the **hackrf_one_usb.dfu** by using this command : 
"dfu-util-static.exe" --device 1fc9:000c --download "hackrf_one_usb.dfu" --reset
(2022 & 2017 version)

They both return the same error : 

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

DFU suffix CRC does not match
A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download        [=========================] 100%        42564 bytes
Download done.
unable to read DFU status after completion
can't detach
Resetting USB to switch back to runtime mode

I'm sorry if I'm doing bad. Just got the HackRF so that's a whole new world for me 😅

NeyZeeN, 2024-01-12:
Hello, 
Just received a new PortaPack H2. It's now **working**. 
For now, I can only use it using the micro USB cable. I tried to plug a battery and it went **white screen** again. Unplugged and now it's still working on micro USB.

NeyZeeN, 2024-01-22:
Update : 
Plugged it again and now works fine, even for new speakers.

D0lby55, 2024-01-23:
hi! did you try what NeyZen made: remove the square battery and plug the hackrfpp with the usb cable to a power source?

Chaminoux, 2024-02-17:
Everything is working with the latest firmware 2.0 (installed from Linux). Thank you, everyone!

closed 2024-02-17 by Chaminoux

Brumi-2021, 2024-02-17:
Hi @Chaminoux , glad to see that it fixed now ! 😀 
I believe that this due to the latest auto LCD driver function PR added by @bernd-herzog, a few days ago. 
In the past , in some cases , after having white screen bricks  I use to need to go back to old fw to recover it , but now, according to  my latest  test ,  the  fw 2.0 should be the recorver solution . 
Thanks for your confirmation feedback !

Chaminoux, 2024-02-17:
Thank you @bernd-herzog and everyone else for the work done, it's the pleasant surprise of the weekend !

Brumi-2021, 2024-02-17:
👏👏👏👍 and in case that you see dark icon colors or not working well the knob rotary encoder , it could be easily solved , just reseting and save the p  memory management in the settings . 
Have a nice weekend 😊