portapack-mayhem/mayhem-firmware #1687: Hackrf usb doesn’t show up

logicbreaks, 2023-12-28:
So basically i didn't use my hackrf for a little while now and the firmware on it is 1.7.4. Recently i decided to use it again and wanted to start with a firmware upgrade, so i plugged into the usb port (os: Linux Mint 21.2 x86_64) of my laptop and the device didn't show up, not even on lsusb, just like dead. The hackrf on itself works fine though, it boots it does what it has to do and it charges, only the usb doesnt work. I already tried several things like plugging into a different pc or using a different cable, nothing worked.The connector doesn't look broken and i currently don't have a multimeter to check if it does at least something.

Ps: I know there already was a issue created about this but there wasn't really a final solution at least not for me.

NotherNgineer, 2023-12-29:
Did you run the "HackRF" app or the "SD over USB" app, or were you just expecting to see it over USB after just plugging it in with no apps running?

If you suspect an issue with 1.7.4 firmware, you could also try updating firmware to the latest 1.9.0 by copying the firmware bin file to the FIRMWARE folder on the SD card and running the "Flash Utility" (as described in the Wiki).

logicbreaks, 2023-12-29:
@NotherNgineer Hi there and thanks for the answer! Yes I did use the hackrf mode and also tried the SD over usb function. I also had the idea to flash the newest firmware over the SD card but I thought it may be too risky in case it crashes somehow or the flashing doesn't work in which case I would completely loose my hackrf. Is it that risky or can I flash without worries?

NotherNgineer, 2023-12-29:
I have never had any issue flashing Mayhem firmware using the "Flash Utility" app with a firmware bin file on the SD card (but you'll definitely want to wait 15 seconds for the LEDs to stop blinking to indicate that the Flash is complete).  I recommend unpacking the entire "COPY_TO_SD_CARD" zip file contents to the SD card since many apps are now "external" (in the APPS folder).  This should get you to the latest firmware, but almost certainly won't resolve the issue with USB.

If you want to try some more experiments, try disconnecting the PortaPack board from the HackRF board and see if you can communicate with the HackRF board itself over USB (check dmesg output, and try the "hackrf_info" command for example).

logicbreaks, 2023-12-30:
@NotherNgineer Thanks for your answer again! I now flashed the newest firmware and it did work fine. Obviously I'm not fully happy as the usb doesn't work and it doesn't look like it ever would again, and I think I know what the source of the problem was, I plugged the hackrf onto a Anker powerbank which probably fried something in there but I'm still always open for new ideas or solutions and I would also like to leave this problem open as it's not really solved. Thanks for your help though.

NotherNgineer, 2023-12-31:
If the USB port on the HackRF is damaged, I suggest checking the USB micro connector for bent pins (the middle 3 pins are the data pins) or try replacing the ESD-protection diode pack LXES1TBCC2-004 (U15 on the PCB).

No firmware workaround for this one.  :-)

logicbreaks, 2023-12-31:
@NotherNgineer I already checked the usb port for damage and there isn't any visible damage on the port itself. And just to be sure by U15 do you mean this thing?: [image]

NotherNgineer, 2023-12-31:
No, I meant the U15 below.  But if you have a different HackRF schematic then please attach it.

[image]

logicbreaks, 2023-12-31:
@NotherNgineer Could you please send me your full schematic or link just to have a better overview? My schematic is a bit hard to follow.

NotherNgineer, 2023-12-31:
Kicad schematics of the latest HackRF r9:
https://github.com/greatscottgadgets/hackrf/tree/master/hardware/hackrf-one

gullradriel, 2024-01-12:
Requalified your issue so it can continue.
Good luck 🤞

closed 2024-03-23 by gullradriel

SychevDenis, 2024-06-08:
Hello everyone. I had a similar problem. It got to the point that I made a brick out of hackrf, but the usb does not work. Studying the circuit and checking everything with a multimeter, I came to the conclusion that I have a defective part U15. I got it out and it worked for me. These are essentially diodes that protect the microcontroller from voltage surges, it burned out and caused problems along the USB signal line. I will look for an alternative to it, while I use it without it

logicbreaks, 2024-06-09:
@SychevDenis Thanks so much for the information I wasn't active for a while but this somehow gives me motivation to try it again (honestly I pretty much gave up my hackrf). If you've got any photos from the process or where the u15 part is located i would really appreciate it. But if not it's ok Im just going to try reading the schematic.

SychevDenis, 2024-06-09:
[image]
I took a picture from the Internet and circled the desired detail in red. You can solder or carefully break it off and everything can work for you. I had a problem with this detail. In a good way, a new one should be soldered in its place, this part is called lxes1tbcc2-004, you can choose its analog. It will also work without it, but there will be no protection against static voltage

logicbreaks, 2024-06-10:
@SychevDenis Thanks very much for the explanation this helps a lot!

logicbreaks, 2024-06-11:
@SychevDenis Hi there again, I now get the part off (I know it's pretty ugly) but do I have to bridge any connections now I mean it connected the data wires in first place right?
[image]

SychevDenis, 2024-06-11:
@logicbreaks Hi, I advise you to wipe it all down with good alcohol. In general, you just need to remove this part and that’s it. You don't need anything extra, just remove it. The main thing is to make sure that all 6 contacts that are in place of the microcircuit do not touch each other.

logicbreaks, 2024-06-22:
@SychevDenis I did everything as said, new problem I have now is if I want to turn it on on battery the screen flashes for a millisecond but turns off immediately idk if I did anything wrong or if some contacts are now touching but it thought I made sure they don't. And the usb still doesn't work. The portaback/hackrf now only turns on if it's connected to power but the problem can be with the battery itself because I charged the whole thing thought the port and the battery indicator on the knob said it's fully loaded. If you or someone else have a clue what happened please give me your ideas.

logicbreaks, 2024-10-15:
@SychevDenis Hi there, I know it's a late response but as I'm still interested in using my device, I want to talk about it. I'm not that kind of a techy I mean, yes I have a multimeter but that thermo thing for I example, I don't. And also I don't have much capital for buying such stuff do you maybe have some low budget solution (Ik that's kind of a dumb question bat anyways)? Or could you maybe offer me to check my device if I would send it to you (of course I would cover the costs of shipment).