portapack-mayhem/mayhem-firmware #1690: Broke during update

utenus, 2023-12-29:
**Describe the issue**
When trying to update the system according to the tutorial, something broke because the portapack does not turn on. The device is not detected when connected to USB in Windows.

**My Hardware**
https://s.click.aliexpress.com/e/_DmU7GQX

**Affected versions**
v1.8

**Were you able to update the firmware before?**
v1.7.4

**Can you try the upgrade with a different PC/Portapack/HackRF?**
No

NotherNgineer, 2023-12-29:
Which firmware update method were you using, and was it 1.8.0 that you were trying to flash?  If the flash operation is interrupted, you may have to attempt a DFU recovery https://github.com/eried/portapack-mayhem/wiki/Won't-boot

Please clarify if the battery has been fully charged by connecting it to a power adapter until the 4 blue LEDs come on (it doesn't charge when attached to a computer USB port in my experience).

utenus, 2024-01-03:
Problem resolved.

closed 2024-01-03 by utenus