portapack-mayhem/mayhem-firmware #1693: Power-up initialization issue affecting 1.8.0 & later

NotherNgineer, 2023-12-29:
I believe there may be a power-up initialization issue affecting the 1.8.0 firmware and later releases.

Starting from a power-off state, I double-click the encoder button which briefly powers the PortaPack on & off.  Seconds later I press the encoder button once to power on.  Result is as follows:

1. Using 1.7.4 firmware and earlier, firmware always starts up properly after the sequence above.
2. Using 1.8.0 firmware and later, the screen stays blank and the top LEDs all blink rapidly until powered off.  In the case of an R9 board, firmware will update Pmem to disable the TXCO following this test sequence.

I have assigned this issue to myself to determine which change introduced this.  In the mean time, the workaround is simply to power cycle.  And, if it's a HackRF R9 board, the user may wish to re-enable the TXCO in Settings->Radio.

jLynx, 2023-12-29:
So you only have this issue when TXCO is enabled?

NotherNgineer, 2023-12-29:
This occurs regardless whether TXCO is configured to be enabled or not.

jLynx, 2023-12-29:
gotcha

kpwg, 2023-12-31:
I have the same problem with H1 and at least 1.9.0 onwards. Pressing RESET once fixes the situation

NotherNgineer, 2023-12-31:
Looks like this issue was introduced in PR #1433 (n230905 nightly release).

jLynx, 2023-12-31:
@bernd-herzog related to our change. Apparently works regardless on weather it's enabled or not

NotherNgineer, 2024-01-01:
It seems that this may be _expected_ behavior when the encoder knob is accidentally pressed twice (starting from an off state), because the CPLD shuts power back off quickly and it looks to firmware like a start-up failure.

I found this helpful Wiki page from @bernd-herzog that explains the blinking "config mode" state that Mayhem firmware enters into following any start-up failure:  https://github.com/eried/portapack-mayhem/wiki/Config-Menu

It is unfortunate that there's nothing on the display to indicate what's happening, and that this accidental double-press of the reset results in the external clock being shut off unnecessarily on r9 boards.  Possible enhancements perhaps?

ETHANATTI, 2024-01-12:
I can't turn off my device. Is it possible?

NotherNgineer, 2024-01-12:
> I can't turn off my device. Is it possible?

Make sure the USB cable is disconnected and double-click the encoder button to turn it off.  The CPLD may need to be programmed if that doesn't work.

NotherNgineer, 2024-01-16:
To work around the original complaint above, perhaps we could add a new setting in persistent memory to disable Config Mode.  (If Config Mode was really really needed after it was disabled then I could pull out the coin cell battery to reset the persistent memory.)

gullradriel, 2024-01-16:
> To work around the original complaint above, perhaps we could add a new setting in persistent memory to disable Config Mode. (If Config Mode was really really needed after it was disabled then I could pull out the coin cell battery to reset the persistent memory.)

Good idea

closed 2024-01-25 by NotherNgineer