portapack-mayhem/mayhem-firmware #1696: Tesla Charging Port Opener from SDCARD not working

pd9mwo, 2023-12-30:
Tesla Charging Port Opener doesn't work from the SDCARD.
But when i use the website version connected to my phone it works !!!

This one works : https://xakcop.com/tesla-opener/
Th version on the SDCARD not ;-(

gullradriel, 2023-12-30:
Not enough informations.
What were the settings ?
Which firm version ?

gullradriel, 2023-12-30:
Your 'phone' or 'phone app' may be setting the correct freq, VGA, LNA, gain, that you have to set yourself in Mayhem.

pd9mwo, 2023-12-30:
Sorry gullradriel for the missing info ..
Files came from the 1.9.1 official firmware
https://github.com/eried/portapack-mayhem/tree/next/sdcard/SAMPLES

I run these on the hackrf + portapach h2+ with the latest 1.9.1 firmware.
Those are not working with the "replay" option .
When i use my phone and browse to that url https://xakcop.com/tesla-opener/  and connect my hackrf to my phone it works

I hope you have enough info now
many thanks in advance
m

NotherNgineer, 2023-12-31:
What kinda of phone is this that you're able to connect a USB cable from the phone to the HackRF to transmit?  Did you compile the HackRF tools on your phone?  Perhaps unrelated to the Replay issue at hand, but I'd like to hear more about that!

pd9mwo, 2023-12-31:
NotherNgineer
i use a samsung flip 5 , nothing special
i copiled nothing on my phone ... just goto the website and you wil see : https://xakcop.com/tesla-opener/

NotherNgineer, 2023-12-31:
That's verrry interesting.  Off topic, but we could probably leverage that javascript to flash HackRF firmware as well...
https://github.com/rgerganov/tesla-opener

gullradriel, 2024-01-02:
@pd9mwo has it ever even worked on your PP ?
-Try using a nightly build
-Have your sdcard content up to date
-Check that your CPLD is up to date
-Reset your Pmem in Settings/PMem
-Clear your settings in SETTINGS
-Clean reboot (power off / power on, not reset)
-Check the AMP/Gain settings you are using are the same than in the webapp

pd9mwo, 2024-01-03:
SUPER THANKS gullradiel
Works !!!

_-Try using a nightly build -> **Done ! Nightly Release - 2024-01-02**
-Have your sdcard content up to date ** Done!**
-Check that your CPLD is up to date **Done used the latest version inthe firmware zip file **
-Reset your Pmem in Settings/PMem **Done**
-Clear your settings in SETTINGS **Done**
-Clean reboot (power off / power on, not reset) **Done**
-Check the AMP/Gain settings you are using are the same than in the webapp **Done*_

gullradriel, 2024-01-03:
You're welcome

closed 2024-01-03 by gullradriel