portapack-mayhem/mayhem-firmware #1718: I was trying to turn on my HackRF and it is a blank screen with the dill flashing.

BeatboxT, 2024-01-05:
----
Before creating this issue, **do the following**: 
* Read the Wiki on booting: https://github.com/eried/portapack-mayhem/wiki/Won't-boot
* Read: https://github.com/eried/portapack-havoc/wiki/Update-firmware
* Watch carefully: https://www.youtube.com/watch?v=_zx4ZvurgOs
* (if you are not in Windows) also check: https://www.youtube.com/watch?v=kjFB58Y1TAo

----

**Describe the issue**
A clear and concise description of what the issue you are facing is. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'

**My Hardware**
Please specify what PortaPack hardware version you are using. 
You can find the list of versions here: https://github.com/eried/portapack-mayhem/wiki/PortaPack-Versions

**Affected versions**
Please tell us what version you are running. 
If your device is still functional, try the latest nightly release before submitting this. 
You can find the latest nightly version here https://github.com/eried/portapack-mayhem/releases

**Were you able to update the firmware before?**
Things might be confusing the first time, please check the video available on the link above.

**Can you try the upgrade with a different PC/Portapack/HackRF?**
If is possible, swap hardware and try again. Also, try different USB cables, even if the one you are using works fine for other purposes.

**Additional**
If the issue is difficult to explain, additionally to the text please include images and videos.

NotherNgineer, 2024-01-05:
Not sure what "dill" flashing means, but if you power it off by pressing the encoder button twice, does it power up correctly the next time the encoder button is pressed?   Perhaps a duplicate of #1693 ?

mike-cca, 2024-01-14:
Hi Mayhem team,
I don't want to start a new thread and I think that the author of this issue meant what is in the attached video.
https://github.com/portapack-mayhem/mayhem-firmware/assets/156543080/35d03e53-e4d7-4bca-a822-fe75c55c60d6

I have been using HackRF & PortaPack for 5 years and I used Sharebrained firmware for PoprtaPack, later I updated to Mayhem v 1.2 and for a long time I stayed that way without problems until version 1.8.0.

jLynx, 2024-01-14:
@mike-cca you are in the bios. @bernd-herzog can you please post the link to it as I can't seem to find it

mike-cca, 2024-01-14:
I realize that the hardware combination I use is not the most used, nor is it the cheapest, but it is very accurate for my use, and has been working without major changes for 5 years.
Pressing the reset button in FW 1.7.4 after connecting is not annoying. For power supply I use a powerbank tested by Fluke scopemeter for power supply faults. Some powerbanks have terrible power supply.

NotherNgineer, 2024-01-16:
@mike-cca  Nice testing report.  Does your PortaPack have an internal battery, or does "booting" in your case always mean turning on or plugging in an external power supply?

It seems that a lot of us are having an intermittent issue since n_230905 where Config Mode is entered intermittently when the USB cable is plugged in as well, which could be related.

mike-cca, 2024-01-16:
I don't have internal battery, I have PortaPack H1 R2 without modifications. You could certainly add a battery, but it would require PCB intervention and it would still be just a hard power on.
I currently use a Li-Ion powerbank without various fast charges, it has a button to turn off and on USB power. I have verified the whole process of turning on HackRF and PortaPack with Fluke Scopemeter and the process is clean without any spikes in the power supply.  Of course it boots every time you turn on the power.

Portapack H2 with built-in functionality for internal battery power supply and integrated charging does not boot at every start ?
Does the PortaPack H2 have any standby mode ?

NotherNgineer, 2024-01-16:
> Portapack H2 with built-in functionality for internal battery power supply and integrated charging does not boot at every start ?  Does the PortaPack H2 have any standby mode ?

@mike-cca  Thanks for the info.  In my case I have two H2+ Portapacks, both of which have the internal LiPo batteries.  When the USB is disconnected, a single press of the encoder button turns it on (seems to always work), and a double press turns it off (it cannot be turned off when USB is connected).  I suppose you could refer to the "off" state as "standby" mode.  But on either unit, if I start from an off state and accidentally double-click the encoder button then it will enter Config Mode instead of starting up correctly using n_230905 firmware or later (noted in issue #1693).  Secondarily, on one of my two units I have an issue that sometimes it will also enter Config Mode when the LiPo is fully charged and I plug in the USB cable, which I am guessing could be due to power glitches when the cable is attached.

The "Config Mode" code also stores a flag in persistent memory to determine how to behave on the next power-up.  Can you tell me if you have a working coin-cell battery (which is used to retain settings in persistent memory)?

mike-cca, 2024-01-16:
@NotherNgineer otherngineer
It is questionable how the standby state is defined from the HW and SW side for the portapack H2.
Does the H2+ boot when powered from the internal battery at one press ? 
I have tested the last firmware and by double pressing the reset button I enter Config Mode also on my HW.
But even with controlled power on via USB, every second attempt is to enter Config Mode and only a few attempts is to boot into the MAYHEM menu.
I have checked the CR2032 battery for persistent memory and time data backup but also replaced it with a new one. I have also tested the backup power supply by measuring if there are any transients or discharges when the power is supplied from the backup battery to the persistent memory. Everything is fine.

NotherNgineer, 2024-01-16:
> Does the H2+ boot when powered from the internal battery at one press ?

Yes

mike-cca, 2024-01-16:
@NotherNgineer otherngineer
I have read almost all the relevant trains and #1693,
It's definitely a common problem. I've tried all the advice in Won't boot.

I'm an electronics engineer for a couple of decades but only a basic programmer :-)
PortaPack H2+ has additionally power and charging circuits of Li-Pol battery and by double pressing the battery is switched off and keeps only the charging circuit and power part but HackRF is off.

Is there a PortaPack H2+ wiring somewhere ?
Maybe there are differences. With no battery the Portapack is powered by the HackRF, and with an internal battery it is the other way around, the PortaPack H2+ provides the power for the HackRF.

NotherNgineer, 2024-01-17:
Attached is a test version of firmware that adds a "Config Mode" app to the Settings menu (see PR #1781).  The app can be run to disable Config Mode via a persistent memory setting (this test firmware version is currently the only one that looks at the persistent memory setting BTW)

[portapack-h1_h2-mayhem.bin.zip](https://github.com/portapack-mayhem/mayhem-firmware/files/13957883/portapack-h1_h2-mayhem.bin.zip)

mike-cca, 2024-01-17:
@NotherNgineer 
Hi, which firmware is modified by adding disabling Config Menu ?
I will test it.

NotherNgineer, 2024-01-17:
@mike-cca  You can unzip the test firmware I attached above and flash it, or use tonight's nightly firmware build n_240118.  Both have support for disabling Config Mode (look for Config Mode app on the Settings page).

mike-cca, 2024-01-17:
@NotherNgineer
I have tested the firmware with Config Mode disabled. This fixes the problem of crash booting directly into Config Mode. It does not fix the unreliable booting itself.

I also tested the Nightly Release - 2024-01-17, with the CR2032 backup battery removed. With the battery in place, booting to Config Mode is invariably with invalid boot. Without the battery, the reset is still invalid boot and Config mode, which is logical because the persistent memory is still powered. When the power is disconnected and reconnected, Config Mode is disabled, which will be caused by the fact that there will be no record of invalid boot in the persistent memory.

mike-cca, 2024-01-17:
@NotherNgineer
Why is there a significant difference in boot reliability in new firmware ?
For example 1.9.1 ( all newer than 1.4.4) as in the video :
https://github.com/portapack-mayhem/mayhem-firmware/assets/156543080/e5a9e7f8-ee73-4ace-910b-aa2fce0c4c09

And this is how HackRF One R4 & Portapack H1 R2 boot in firmware version 1.4.4, absolutely 100% :
https://github.com/portapack-mayhem/mayhem-firmware/assets/156543080/12b93004-fc53-4896-9dbd-4775ac0bd7f6

NotherNgineer, 2024-01-17:
If an issue started occurring in a specific release, it would help if someone could test the nightly releases leading up to that, to determine the precise date when the issue was introduced.

mike-cca, 2024-01-17:
The last version that boots with my HW at 100% is Nightly Release - 2022-04-05.
Nightly Release - 2022-04-06 is the first release when HackRF One R4 & PortaPack H1 R2 does not boot reliably.

mike-cca, 2024-01-18:
A big thank you to the developers for the ability to disable Config Mode in the Nightly Release menu - 2024-01-18.

NotherNgineer, 2024-01-18:
> @NotherNgineer The last version that boots with my HW at 100% is Nightly Release - 2022-04-05. Nightly Release - 2022-04-06 is the first release when HackRF One R4 & PortaPack H1 R2 does not boot reliably.

As you try out nightly versions to determine the point where the instability began, keep in mind that there might be multiple ranges of version that work or don't work, as issues get resolved and new ones introduced.

I reviewed the code changes in the 2022-04-06 nightly release and it looks mainly like some of the power-up code was reordered, which could possibly have introduced a timing issue.  I'll keep thinking about it.

mike-cca, 2024-01-18:
Thank you for your time.
Yes there may be a problem in the timing, that's why the correct boot is random.