portapack-mayhem/mayhem-firmware #1724: How to use apps / external apps ?

h0us3c4ll, 2024-01-05:
hi, i have firmware v1.9.1.

all files copied to microsd.

but how do i access external apps, for example jammer? if i check file manager, i see it in /APPS, as a .ppma files.

thanks

gullradriel, 2024-01-05:
Be sure that all files/dir are at the root of the sd card, and not in a "sdcard" directory at the root of the sd card itself

NotherNgineer, 2024-01-05:
To clarify what gull mentioned, if you open up the File Manager app it should look similar to this:

[image]

loewal, 2024-01-08:
Same here..
I copied the contents of the latest "copy to sdcard.."
Then tried to start an external program: "the ppma file in your APPS folder is outdated".
Why? It is the latest....?

gullradriel, 2024-01-08:
@loewal the external app pack have to match the firmware bin. So you have to update both, flash bin and copy apps to sdcard.
@h0us3c4ll if they are copied correctly, you'll see more apps in the interface. Nothing to see in filemanager except that the file are there or not.

h0us3c4ll, 2024-01-08:
ah, i see now, thanks for support !

closed 2024-01-08 by h0us3c4ll

Smonyer, 2024-03-09:
Just downloaded new v.2.0 to portapack, copied files to SD card correctly to root . I have no ppma file downloaded and can't unzip new one ,,mayhem_v2.0.0_OCI.ppfw.tar  ,,, to SD card correctly for apps to work,it keep saying no external apps , pleas help,, thanks

gullradriel, 2024-03-09:
You have to also flash the corresponding firmware provided with it.
Do not unzip manually.
Copy the ppfw.tar file on your Sdcard and use the utilities/flash app onboard

Smonyer, 2024-03-11:
I got it thanks , still trying to figure out how to get the world maps  app
on there ?

Kingdamienjl, 2024-11-18:
so the .cpp and .hpp in C:\Users\Pimpn\Downloads\mayhem-firmware-next\mayhem-firmware-next\firmware\application\apps  
are not additional apps to add to the apps folder on sd right?

gullradriel, 2024-11-18:
> so the .cpp and .hpp in \mayhem-firmware-next\mayhem-firmware-next\firmware\application\apps are not additional apps to add to the apps folder on sd right?

No, you should not copy that. What you are seeing here are the source code files.

Kingdamienjl, 2024-11-18:
so i guess what my real question is, are there anymore apps aside from sdcard i should or could be using? im used to a flipper zero amount of apps lol..

gullradriel, 2024-11-18:
> I got it thanks , still trying to figure out how to get the world maps app on there ?

There is no "World Map app". There is only a worldmap database for ADSB app and a few others.

As I said before, to prepare your sdcard:
- first unzip sdcard content from the release page at the root of your sdcard
- then update using hackrf.app or putting the ppfw file on your portapack and use the flasher in "utilities". Both these methods are updating the external apps on your sdcard

Kingdamienjl, 2024-11-18:
ok so then no, the only apps available are all on the mayhem giuthub. thank you for your help.

fkarlitepe, 2024-11-26:
> hi yes, sorry, should be clearer. i see same file structure, so that is OK. but then how to use the app itself? if i select it, nothing happens...

I filled the SD card as requested. v2.0.2
then I did the v2.0.2 update, but the applications Jammer etc. do not appear. Please help.

gullradriel, 2024-11-26:
The apps are mostly in the submenus, and at the end of the listing in each submenu.

fkarlitepe, 2024-11-26:
Which submenu do you think it is in?
[image]

gullradriel, 2024-11-26:
Applications should indeed be here and are not appearing.
I don't see why.
Please try to update using the hackrf.app website (it's updating firmware and sdcard external apps)
or
Download and copy the [ppwf file](https://github.com/portapack-mayhem/mayhem-firmware/releases/download/nightly-tag-2024-11-26/mayhem_nightly_n_241126_OCI.ppfw.tar) from the [release page](https://github.com/portapack-mayhem/mayhem-firmware/releases), and use Utilities/Flasher to flash it on the device (it's also updating  firmware and external apps)

Kingdamienjl, 2024-11-26:
Man's might have a clone rf

gullradriel, 2024-11-26:
Should work with any H1/H2/H2M/H4M and hackrf original or clone.
The only unsupported device is the H3, I can't say if it works or not on it.

fkarlitepe, 2024-11-26:
thank you so much. when it is updated, it is perfectly working

h0us3c4ll, 2025-03-24:
hello, sorry for ask again , now with website https://hackrf.app/, do you need seperate to upload the apps ? or can firmware and apps be done via website ?

thanks

gullradriel, 2025-03-24:
> hello, sorry for ask again , now with website https://hackrf.app/, do you need seperate to upload the apps ? or can firmware and apps be done via website ?
> 
> thanks

If you update using the website, it's updating both firmware flash and the external sdcard apps. You have to have a valid sdcard in your unit.

Eltiefao, 2025-07-12:
My RF Porta Pack H2 hack is not showing the ADSB maps since I bought it. I have never been able to view the map. I have already put all the files and nothing appears. When I put the files on the SD card, I cannot transfer the maps.bin. If it does not give an error in the RF hack, I have to put the SD first without this file. Turn on the RF after that, when I put the file, it is already free to use the others. The maps do not appear.

gullradriel, 2025-07-12:
> My RF Porta Pack H2 hack is not showing the ADSB maps since I bought it. I have never been able to view the map. I have already put all the files and nothing appears. When I put the files on the SD [...]

At first glance I would say either or both the sdcard and sdcard reader are dead. Try to change sdcard. Check that it's FAT32 or extfat.

Then go on the release page and get the corresponding sdcard package, not the one without world map, but something like this one:[mayhem_v2.2.0_COPY_TO_SDCARD.zip](https://github.com/portapack-mayhem/mayhem-firmware/releases/download/v2.2.0/mayhem_v2.2.0_COPY_TO_SDCARD.zip). It should have a size around 546 MB.

Unzip it AT THE ROOT OF YOUR SDCARD, all the directories inside the zip have to be at the root of the sdcard, like in that picture (it's only the few first):

[image]

Maybe you should read the wiki, watch some youtube videos (like this one https://youtu.be/n7N7G_9tj9w?si=J4idyD6vllf1GZe0)

Also, please open you own issue.