portapack-mayhem/mayhem-firmware #1745: Still ADSB-X Error ? No planes seen

execccuter, 2024-01-08:
I still cannot use adsb-x, see no plains no map 

I tried to send a signal first on another freq then check the map / the adsb nothing incoming ...
 

cant fix?

Brumi-2021, 2024-01-09:
Hi which fw version are you testing ?
Did you try with the last nightly ?

execccuter, 2024-01-09:
Hello, Im using the latest Firmware? 1.9.1?--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

eried, 2024-01-10:
It might be possible that your amp is blown? Can you receive other signals?

execccuter, 2024-01-11:
rx_adsb is the correct name i delete it now

execccuter, 2024-01-11:
only the rx or the tx too?

execccuter, 2024-01-11:
> try to delete adsb_rx.ini in the "settings" folder , and restart the app! this worked on the 1.9.0... you can find the topic about this in the "closed" section.

still nothing to see ...

gullradriel, 2024-01-11:
- Flash the nightly
- Reflash your CPLD if not sure if it is up to date
- Clear PMem Settings
- Delete all files in SETTINGS directory
- reboot
- should be good

D0lby55, 2024-01-11:
i do not use the tx not legal where i live...

when you are not sure and don't want to loose the file,add  ".old" after the extension, like this "rx_adsb.ini.old" to not loose it.
this files are rebuilt when you launch the app.