portapack-mayhem/mayhem-firmware #1746: Suggest Bag to fit Portapack H2 Kit

PowerPress, 2024-01-09:
Can anyone suggest a bag that will hold all HackRF Portapack with kit from

https://s.click.aliexpress.com/e/_DmU7GQX

TY

eried, 2024-01-10:
I would guess a good fit is a 3.5 inch hdd case like https://www.aliexpress.com/item/32844596844.html but I guess you want to have some other antennas and accessories, so if you feel collaborative I would suggest to kickstart a wiki entry about "carrying cases" where some options could be added 😁

PowerPress, 2024-01-10:
Its all the antennas you get in the kit some 11.25 inches make it more complicated

eried, 2024-01-15:
I dont think this is an issue of the fw itself, @PowerPress please if you find something create a wiki entry as I commented you or discuss in discord :)

closed 2024-01-15 by eried