portapack-mayhem/mayhem-firmware #1751: Pocsag no scroll down

f1ghy, 2024-01-10:
Hi, On pocsag app: When there is to much meesages  for the screen, you can't scroll down, as on the other app. so you have to look at the logs to see the last messages.
Thanks ;o)

Go to pocsag app
   wait there is too much messages to see all of them,
      try to scrwl down to see the last messages: youy cant

D0lby55, 2024-01-11:
this is the right way to read all the received messages by looking in the "log" file! you can’t scroll down with this app...she push the last received message.

gullradriel, 2024-01-12:
I have replaced that issue as an enhancement request.
PR's are welcome 🤗
Thanks ;-)

f1ghy, 2024-01-12:
Thanks, that is not an emergency as a bug, you are right ;o)