portapack-mayhem/mayhem-firmware #1752: Recon guru when select input/output file

f1ghy, 2024-01-10:
Hi,
On recon app, in config menu, when try to select input file, or output file... then Guru meditation:

Go to Recon app,
   go to config menu,
      Hilight "Select input file"
          press the center buton... then guru meditation

Go to Recon app,
   go to config menu,
      Hilight "Select output file"
          press the center buton... then guru meditation
Thanks

gullradriel, 2024-01-10:
Not an issue and not reproducible on my 3 devices, because it's what it's saying, a memory over usage, but it's not in Recon since it's after a click on input/output. The Guru is coming from freqman explorer.
You may have too much files in FREQMAN directory and the File Manager can not load it, not on top of Recon.
Make some cleanings and it should go away :-)
I'm waiting for your confirmation to close the issue.

NotherNgineer, 2024-01-10:
Happened to me too when I had 65 files in the FREQMAN directory.  Note there are 57 files in the directory in the current SDCARD image on Github, so I guess it only takes a couple more before this happens.

gullradriel, 2024-01-10:
We should make freqman handle the full path instead of a single one.
Sorting with subdirs would greatly help

gullradriel, 2024-01-10:
I think we discussed it at some point with @kallanreed

f1ghy, 2024-01-10:
SHAME ON ME !!! How could i not think about it to myself !!!
Of course, you are right ! I have clean FREQMAN dirctory... now it's ok !!!
Thnaks for the help !!!
;o))

closed 2024-01-10 by f1ghy

NotherNgineer, 2024-01-12:
Unfortunately this issue just gets worse as more features are added to those apps that use File Manager so we should come up with something.  In my case, the FREQMAN directory that had 65 files worked fine with Recon in firmware version 1.9.0, so I suspect that the "repeater" functionality put it over the edge.