portapack-mayhem/mayhem-firmware #1755: Is there actually an acoustic s-meter for fox hunting?

hem0773, 2024-01-11:
Is there actually an acoustic s-meter for fox hunting?

NotherNgineer, 2024-01-11:
As the receiver/hunter, you could try the Level app to view signal levels & listen (please suggest specific improvements if it doesn't work out).

(BTW, when acting as the transmitter/fox, the Morse app is suggested).

htotoo, 2024-01-11:
On my roadmap is to implement a specific external app for fox hunting, based on Level app, so I'll implement your suggestions into it (too).

hem0773, 2024-01-11:
Thanks, I thought of something like that
https://github.com/portapack-mayhem/mayhem-firmware/issues/371

gullradriel, 2024-03-23:
You now have beep available in Level app

gullradriel, 2024-03-24:
You now have beep available for ranges as well in Looking Glass app

closed 2024-04-06 by gullradriel