portapack-mayhem/mayhem-firmware #1764: ADS-B - Cant see the map, only black

Nirwnana, 2024-01-12:
I have the problem that the map view shows only black instead of the map. i tried the nighty build and the stable build 1.9.1
[image]

NotherNgineer, 2024-01-12:
Please verify that the world_map.bin file size is 2.1GB.  This might happen if the bin file was not generated completely from the jpg file.

Nirwnana, 2024-01-12:
just copied from the original build, did not generate myself

[image]

Nirwnana, 2024-01-12:
It seems like they have the exact same problem as me, unsolved:

https://www.reddit.com/r/hackrf/comments/176gh8r/portapack_h2_mayhem_v174_adsb_world_map_will_not/

NotherNgineer, 2024-01-12:
Can you also check if the map works in other apps such as Radiosonde?  (In Radiosonde, click "See on Map" and then increase the latitude value.)

You could also try zooming in/out to see if anything appears.  To zoom the map, move the cursor down to the map area by pushing the down key several times, and then turn the encoder dial clockwise to zoom in, or counter-clockwise to zoom out.

eried, 2024-01-15:
@Nirwnana as @NotherNgineer asked, if you dont see the file in File Manager-> it might be a hardware fault (sd slot, does soldering seem ok? maybe there is visible bridging) or in the sd itself-> test another microsds

Nirwnana, 2024-01-15:
The new SD card works fine, so it was actually a defective SD card. I guess a 2GB card with firmware was modified as 4GB and data was corrupted

closed 2024-01-15 by NotherNgineer