portapack-mayhem/mayhem-firmware #1772: Receiver gaps

vk2kyp1, 2024-01-16:
I have a HackRF with a H2 portapack  and I have found the following gaps in the receiver, using the sound App.
945 to 1999 MHz
2171 to 2300 MHz
2701 to 2920 MHz
4201 to 4720 MHz

I am using a Sytron Donner 1720 Sythesiser with the output set to -70dBm.
Mayem version 1.8 of the firmware
 The gaps are sharp and I feel the sythesiser is not being programmed correctly?
Thank you
regards Gary vk2kyp

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/portapack-mayhem/mayhem-firmware/releases)

----

**Describe the bug**
A clear and concise description of what the bug is. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'
3. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:
* Latest Nightly release:
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

NotherNgineer, 2024-01-16:
Can you please try the latest Mayhem firmware?  I don't seem to have this issue on my Chinese clone H2+/R9.  ADS-B RX wouldn't work, for example, if it couldn't receive 1090MHz.

vk2kyp1, 2024-01-20:
Hi, Thanks for suggesting a firmware upgrade now running 1.9.1.
1090 and 1296 Mhz are now working on the sound and signal generator apps
Regards gary vk2kyp

NotherNgineer, 2024-01-20:
@vk2kyp1  Are you still unable to receive the other frequency ranges?

vk2kyp1, 2024-01-20:
Hi I have not had time to check the other frequencies, hopefilly later today

vk2kyp1, 2024-01-23:
Hi I have just checked for receiver gaps with the 1.9.1 Mayhem firmware these are:

2171 to 2170 MHz
2700 to 2936 MHz
4201 to 4560 MHz

The HackRF looses sensitvity on the higher frequencies, I would like to hear from anyone to confirm my results

Thank you
Gary vk2kyp

NotherNgineer, 2024-01-23:
@vk2kyp1  Is there a typo in the first one 2171-2170?

I confirm that two PortaPacks cannot communicate with each other on some frequencies.  However, I do not know yet if it's a transmit or receive problem (or both), and if only R9 is affected.

vk2kyp1, 2024-01-23:
I am sorry for the Typo,
the results should be
2171 to 2300 MHz
2700 to 2936 MHz
4201 to 4560 MHz

thank you for spotting this

gary

NotherNgineer, 2024-01-25:
There is certainly an issue, and it seems to affect other HackRF board models as well.

For example, when I set the Audio app for WFM, 200K BW, at 2170.8 MHz, I see a little birdie or something around 2170.9.  But the birdie does not move as I increase the frequency from there, until I get to 2300 MHz.  But then if I turn the frequency down from 2300 Mhz, then whatever it was receiving at ~2301 MHz continues to be received at all frequencies until I get down to ~2170.8 MHz again.

Also, if I manually enter a frequency that has a strong signal such as 2450 MHz, and then manually enter a frequency between 2171-2300 MHz, it seems that the tuner is still receiving the 2450 MHz signal.

So it would seem like setting the frequency to a value in the ranges you've mentioned does not result in any tuning change of the radio receiver, at least within a certain band.

It might be interesting to test this in HackRF mode versus running Mayhem firmware.

NotherNgineer, 2024-01-26:
Just looking at the birdies, it doesn't look like this issue occurs in HackRF mode, when HackRF is attached to a computer running CubicSDR.

NotherNgineer, 2024-01-27:
Please re-test this after nightly firmware n_240128 comes out later tonight.  PR #1819 should fix at least some of these dead bands.

gullradriel, 2024-03-10:
Close due to fix and inactivity of op

closed 2024-03-10 by gullradriel