portapack-mayhem/mayhem-firmware #1773: Location of DJI Drone on map (like ASD-B)

krisss117, 2024-01-16:
**Is your feature request related to a problem? Please describe.**
Add feature like ASD-B, but with location of DJI Drones

**Additional**
https://www.youtube.com/watch?v=4F0T7_sB_Ss
https://www.reddit.com/r/dji/comments/11g89ln/this_hacker_tool_can_pinpoint_a_dji_drone/

htotoo, 2024-01-16:
Souces for apps that decodes this:
https://github.com/proto17/dji_droneid
https://github.com/RUB-SysSec/DroneSecurity

Interesting data in packet:
lat, lon, alt, height, 
lat_home, lon_home,
dev_type, serial

Have to mention, that new DJI's didn't recognized by this, since they communicate in a different way (based on internet, not on own tests)

cprasmu, 2024-01-20:
Not sure the hardware will support the bandwidth required?

fagci, 2024-01-21:
15.36MHz as I understood

[image]

cprasmu, 2024-01-21:
ah ! - I totally misread the gnu radio graph! Think this is an awesome idea if it can be done.

tomventa, 2025-06-01:
In the 2025 edition of nohat (italian conference on cybersecurity) there was a tak titled "not the drone you're looking for" about it. the slides are still available online. This may be helpful in some way