portapack-mayhem/mayhem-firmware #1774: SubGHZ garage protocols (NICE/CAME/FAAC)

krisss117, 2024-01-16:
**Is your feature request related to a problem? Please describe.**
Please add support garage protocols like NICE/CAME/FAAC. Scan, decode, generate new codes and allow to send (like Flipper Zero). 

**Additional**
You can get code from https://github.com/Flipper-XFW/Xtreme-Firmware/tree/dev/applications/main/subghz

htotoo, 2024-01-16:
Here are the supported protocols:
https://github.com/portapack-mayhem/mayhem-firmware/tree/next/firmware/baseband/fprotos

If you set the amp/lna/vga too high or too low, it won't be recognized. Ideal values around 0, 24, 30. Play with it a bit.

krisss117, 2024-01-16:
When I read signal, then SubGHZ show me serial FFFFFFFF and 52 bit NICE - but cant generate new code and send it. Try different values to read - but still the same.

htotoo, 2024-01-16:
Ohh, I see. Sending is not in the roadmap yet. You can use the record and replay with not rolling code ones.

closed 2025-10-17 by htotoo