portapack-mayhem/mayhem-firmware #1779: PPM correction introduces mirroring

fagci, 2024-01-16:
**Describe the bug**
When using PPM correction, signal doubles on waterfall. Frequency correction in Hz work only on single band.

**To Reproduce**
Tune PPM

**Expected behavior**
Single signal persists, but offsetted.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
1.8.1
* Latest Stable release: 1.9.1
* Latest Nightly release: 2024-01-16

**Additional**

https://github.com/portapack-mayhem/mayhem-firmware/assets/8795970/1d305846-f40a-4f75-947a-2b8d663a31e8

gullradriel, 2024-01-16:
Same recipe as usual:
- update your cpld
- flash the nightly
- clean PMem settings
- delete all settings files in SETTINGS
- reboot
- check in settings/radio that external TXCO isn't disabled

gullradriel, 2024-01-16:
You'll maybe have to use your boot key at first boot, and set back your preferences in settings

gullradriel, 2024-01-16:
1.8.1 wasn't our best release because of compiler version problems IIRC

fagci, 2024-01-16:
Thank you! Cleaned settings and "check in settings/radio that external TXCO isn't disabled" solves problem. Now frequency tuning precise as atomic clock =P

closed 2024-01-16 by fagci

gullradriel, 2024-01-16:
Perfect 👍
Have fun !