portapack-mayhem/mayhem-firmware #1785: TPMS for Toyota

ifuchs, 2024-01-17:
Is TPMS for Toyota vehicles supported by the TPMS Cars app? If not, could it be added? (rtl_433 has the code to decode these sensors).

NotherNgineer, 2024-01-18:
It's difficult to add TPMS support for sensor types that we don't happen to have on our own vehicle, but if you could run the rtl_433 app with the -vvv verbose setting and capture the output it might help, so we can see the packet format.  Another possibility is that you could use the Capture app to capture a packet and attach that (if you can get the TPMS sensor to transmit at will, such as by changing the tire air pressure rapidly).

ifuchs, 2024-01-18:
Thanks. I can do that, but wouldn’t looking at the rtl_433 source for
decoding the packet be even more useful?

NotherNgineer, 2024-01-18:
That's probably enough information in the rtl_433 code to create a test version, but the more we have to work with the better.  A capture file would also allow us to transmit the TPMS code on one Portapack to test if another Portapack receives it (theoretically rtl_433 should also show it if you Replay the capture).

u-foka, 2024-02-02:
Hmm, I have a toyota and a kia both with radio sensors... I'm too noob for the radio stuff to implement it at the moment but I can try to capture and isolate their signals if that helps :)