portapack-mayhem/mayhem-firmware #1788: Measurement system selection

fagci, 2024-01-20:
**Is your feature request related to a problem? Please describe.**
ADS-B flight height is in "ft", for me better is meters.

**Describe the solution you'd like**
Setting of distance, temperature, speed.

NotherNgineer, 2024-01-20:
Good idea.  The map shows distance in km, so there's certainly some inconsistencies.  Some apps such as TPMS and Weather allow you to choose between SAE & Metric units, but perhaps this should be a device-level setting.

jLynx, 2024-01-22:
I did the TPMS one. If I get some time I can do this too, but if anyone else wants to in the meantime, go ahead

NotherNgineer, 2024-01-27:
Do we want configurable units in every app (as in TPMS/Weather/etc), or a Settings -> Units app that would allow the user to select the preferred units for each of these (affecting what's displayed in _every_ app)?:

- Pressure (tire):  kPa / PSI
- Pressure (atmospheric):  millibars / inches_Hg
- Temp:  C / F
- Distance/Altitude:  meters-kilometers /  miles-feet
- Speed:  km-per-hour / miles-per-hour / knots
- Weight?:  gram-kg / ounces-pounds
- Volume?:  liters / gallons-quarts-pints-cups-fluid_ounces
- Data storage?:  megabytes (MB) / mebibytes (MiB)

I assume that we _don't_ need any choice of units for these:
- Humidity:  percent
- Time:  Days-hours-minutes-seconds
- Frequency & Bandwidth:  GHz-MHz-KHz
- Sound/signal amplitude:  dB

fagci, 2024-01-27:
I will vote for settings, but maybe in each app selection of roundind to thoudands (ex.: kilometers).

Fathoms in depth? Think its same system, as distance will be.

zxkmm, 2024-01-27:
I’ll vote for different unit set for different app.  Cuz different field use different unit as regular.

jLynx, 2024-01-27:
I think we should have defaults in setting but make it so we can override in the apps. For instance, the I prefer PSI for TPMS when though the metric one is kPa (and I'm from a metic country)

jLynx, 2024-01-27:
Or allow us to configure each item, like temperature, pressure, etc in setting. Then we won't need to override in the app if we can specify the specific unit we prefer to show

gullradriel, 2025-06-22:
That's way too big of a feature, that will touch a lots of firmware parts and cost a lot of firmware space.
Honestly, it will never be done. 
And anyway, everyone should use the International System of Units. It's would cost a lot less to fix the few apps or fields that are using non standard and fucked up measurement systems.

closed 2025-06-22 by gullradriel