portapack-mayhem/mayhem-firmware #1789: Show Getting All Options in Mayhem 1.9.1

seeklakhan, 2024-01-20:
I recently updated to Mayhem 1.9.1 but its not showing all options in Receiver and Transmit Sections please see the screenshots for further [reference]([url](
[image]
[image])) 
  
----
Before creating this issue, **do the following**: 
* Read the Wiki on booting: https://github.com/portapack-mayhem/mayhem-firmware/wiki/Won't-boot
* Read: https://github.com/portapack-mayhem/mayhem-firmware/wiki/Update-firmware
* Watch carefully: https://www.youtube.com/watch?v=_zx4ZvurgOs
* (if you are not in Windows) also check: https://www.youtube.com/watch?v=kjFB58Y1TAo

----

**Describe the issue**
A clear and concise description of what the issue you are facing is. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'

**My Hardware**
Please specify what PortaPack hardware version you are using. 
You can find the list of versions here: https://github.com/portapack-mayhem/mayhem-firmware/wiki/PortaPack-Versions

**Affected versions**
Please tell us what version you are running. 
If your device is still functional, try the latest nightly release before submitting this. 
You can find the latest nightly version here https://github.com/portapack-mayhem/mayhem-firmware/releases

**Were you able to update the firmware before?**
Things might be confusing the first time, please check the video available on the link above.

**Can you try the upgrade with a different PC/Portapack/HackRF?**
If is possible, swap hardware and try again. Also, try different USB cables, even if the one you are using works fine for other purposes.

**Additional**
If the issue is difficult to explain, additionally to the text please include images and videos.

fagci, 2024-01-20:
Are you copied APPS folder tp sdcard?

seeklakhan, 2024-01-20:
where i can find this "..." ?

zxkmm, 2024-01-20:
Hello, please read the wiki here  https://github.com/portapack-mayhem/mayhem-firmware/wiki/Update-firmware  , thanks.

jLynx, 2024-01-20:
From the release page, make sure you download the SD card zip folder and copy that onto your sd card as that contains most of the apps.

closed 2024-01-20 by jLynx