portapack-mayhem/mayhem-firmware #1790: odd crashing problem

Snifi0, 2024-01-20:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/portapack-mayhem/mayhem-firmware/releases)

----

**Describe the bug**
everything seems to work, but when i select receive > audio, or when i press back, or transmit > soundboard, SOMETIMES it just does not show the waterfall, or freeze where i have to power off and on again, and even when it seems to work, it wont transmit and i have to power off and on and go to soundboard till my other sdr picks it up, and it takes like 5 minutes to actually make transmitting work, i have no idea what the problem could be.. it boots and all, but sometimes pressing sound to listen to lets say fm radio, it sometimes freezes without even loading the menu, sometimes it loads the menu but no waterfall. and if its still responsive, when i press back it will completely freeze/crash. the screen just frozen, gotta power it off and on and repeate till everything works, did a [Full reset](https://github.com/portapack-mayhem/mayhem-firmware/wiki/Completely-reset) tried Updating the Xilinx CPLD on hackrf board. not sure what to do.
first used newest stable release and now using newest nightly n_240118

per: RFFC5072, MAX2837, si5351C, WM8731

portapack h2+ from open source sdr lab

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'
3. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release: 1.9.1
* Latest Nightly release: n_240118
* Previous working release: ?

**Additional**
sometimes it will work, sometimes it will freeze.

NotherNgineer, 2024-01-20:
I would recommend trying some older releases to determine if a problem was introduced recently in the firmware.  If the issue occurs with every version, then you might have a hardware issue.  You might also try a different SD card just in case the freezing is associated with file I/O.

Snifi0, 2024-01-21:
> I would recommend trying some older releases to determine if a problem was introduced recently in the firmware. If the issue occurs with every version, then you might have a hardware issue. You might also try a different SD card just in case the freezing is associated with file I/O.

1.6.0 does not want to boot, and newer does and still has the problem. tried 2 sd cards and without and still nothing,

BuddyPage, 2024-01-22:
I was also having similar behavior with regular freezes after moving from 1.8.0 to 1.9... Going back to 1.8.0 seemed to fix most problems. Couple days ago I updated everything and again the same problem, including SD card errors. I was using a 64 GB SD card, which i had installed at the same time I upgraded from 1.8. to 1.9 the first time. I downsized to 16 GB today and most of those problems stopped. Still a little buggy it is much better after downsizing from 64 GB. I think I have an H2/R4, with the black metal case, internal speaker, blue lights on the top of the dial knob, etc.

Snifi0, 2024-01-22:
> I was also having similar behavior with regular freezes after moving from 1.8.0 to 1.9... Going back to 1.8.0 seemed to fix most problems. Couple days ago I updated everything and again the same problem, including SD card errors. I was using a 64 GB SD card, which I had installed at the same time I upgraded from 1.8. to 1.9, the first time. I downsized to 16 GB today and most of those problems stopped. Still a little buggy but it is much better after downsizing from 64 GB. I really worked it today and it was well behaved, mostly. I think I have an H2/R4, with the black metal case, internal speaker, blue lights on the top of the dial knob, etc. 

What about no sd card? Do you get constant crashes still?

gullradriel, 2024-01-22:
I have the same problem on my H2R2 only.
H2R9 and H1R2 OK

BuddyPage, 2024-01-22:
Good question. Didn't try it because my first solution did the trick and my time was short. Also, both sd cards were brand name (64gb Samsung, 16gb Sandisk). I suspect a 32gb would have worked too, but 16 is what I had available. I knew I was pushing it with 64gb, but had to find out...

Snifi0, 2024-01-22:
> Good question. Didn't try it because my first solution did the trick and my time was short. Also, both sd cards were brand name (64gb Samsung, 16gb Sandisk). I suspect a 32gb would have worked too, but 16 is what I had available. I knew I was pushing it with 64gb, but had to find out...

Would you mind trying for me please? Just take the sd put and play for a wicked second I need to know if I gotta get another nother one.

gullradriel, 2024-01-24:
Link to pictures of my buggy device:

https://discord.com/channels/719669764804444213/956561375155589192/1199786330180292758

BuddyPage, 2024-01-24:
Got a link to where it was purchased or an image elsewhere? Can't see this on Discord and tired of trying...

Snifi0, 2024-01-25:
> > Would you mind trying for me please? Just take the sd put and play for a wicked second I need to know if I gotta get another one.
> 
> Took the sd card out and ran through the paces and the only time I could force an error is to do sd card related tasks. I also swapped out the 16 GB and went to 32 GB and so far it seems more stable than the 16gb, which seemed to work fine yesterday, but gave me some write test errors that the 32 GB didn't. I have other minor problems to debug, but my buttons and interface are no longer freezing regularly.
> 
> Also, on the latest update, I setup hackrf tools and flashed the HackRF firmware onto the unit, and updated the Xilinx CPLD, made sure it was working as a HackF, and then flashed the latest PortaPack version (1.9.1). Then I proceeded to troubleshoot the problems that led to downsizing the SD Card.
> 
> Hope that helps.

Thanks, I can say it's not my sd card then...

jLynx, 2024-02-22:
This should have been fixed in v.2.0.0

D33zN0tz84, 2024-09-09:
Disable high speed I-O on sd card setting. This worked for me. when i wanted to capture a signal it started but I could not end the capture.
It states also "this may or may not work"