portapack-mayhem/mayhem-firmware #1796: Add message regarding missing apps

NotherNgineer, 2024-01-21:
As apps are being moved to external files to make room for new features, users are continually reporting that apps are "missing" following a firmware upgrade.  The Wiki mentions copying apps into the APPS folder in several places, but many users don't take the time to RTFW.

Perhaps we need to display a message on the screen, at least once following a firmware upgrade, advising users to copy the external apps to the APPS folder of the SD card.

I'm trying to decide whether this message should appear at every power-up when the APPS folder is missing/empty, or just once following a firmware upgrade.  Opinion anyone?

Assigning this to myself unless someone gets to it first.

gullradriel, 2024-01-21:
I think just once after upgrade may be better, because it will allow to keep some ext apps masked and not being bogged by it at each reboot

zxkmm, 2024-01-21:
not sure if you like this idea? i can do pr for this
https://discord.com/channels/719669764804444213/956561375155589192/1198672424120615023

closed 2024-01-21 by NotherNgineer