portapack-mayhem/mayhem-firmware #1800: Changing frequenty from the clockin port?

pe2ben, 2024-01-21:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/portapack-mayhem/mayhem-firmware/releases)

----

**Describe the bug**
Changing frequenty from the clockin port?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'
3. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:
* Latest Nightly release:
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

pe2ben, 2024-01-21:
Can i change the external clkin reference source frequenty?

NotherNgineer, 2024-01-22:
Please see:  https://hackrf.readthedocs.io/en/latest/external_clock_interface.html

Mayhem firmware currently does not support any CLKIN frequency other than 10MHz.

closed 2024-01-22 by gullradriel