portapack-mayhem/mayhem-firmware #1804: cant put wav files onto the portapack

pduud, 2024-01-22:
**Describe the bug**
wav files disappear in the wav folder in the portapack. using v1.9.1, ive also tried using wav files from the "copy to sd" wav folder.

**To Reproduce**
1. insert sf card into pc
2. insert a wav file into the wav folder on the sd card
3. eject sd card and insert sd card into portapack
4. check the wav folder on file manager on the portapack
5. wav file not there

**Expected behavior**
wav file should be there

htotoo, 2024-01-22:
Should be there. 
What app you use on pp to check it is not there? The file manager?

gullradriel, 2024-01-22:
Bad sdcard or badly formatted.
Have to be Fat32 , not ext Fat32 I.e

pduud, 2024-01-22:
> Bad sdcard or badly formatted. Have to be Fat32 , not ext Fat32 I.e

i formatted with fat32, but the sd card might be offbrand so ill try a known-brand one.

pduud, 2024-01-22:
using a well branded SD card worked, the sd card i got with the portapack is incompatible.

closed 2024-01-22 by pduud