portapack-mayhem/mayhem-firmware #1808: Portapack Crashing due to low battery / cant charge portapack

pduud, 2024-01-23:
**Describe the issue**
i cannot charge the portapack, I plug it into a usb port and the bottom right knob led flashes top signifies the portapack is under 25%, i also hear a quiet tone coming from the portapack. After a while I unplug it and it is still under 25%, i boot it and after a few seconds it crashes. it can boot again but crashes shortly

**To Reproduce**
Steps to reproduce the behavior:
1. Plug in portapack into power brick
3. wait couple hours
4. portapack is still under 25%
5. boot up portapack
6. portapack crashes after around 30 seconds

**My Hardware**
PortaPack H2 Hackrf One 

**Affected versions**
Nightly release - 2024-01-22

**Were you able to update the firmware before?**
Yes

NotherNgineer, 2024-01-23:
I have two PortaPacks myself and neither one will charge when plugged into a computer USB port.  To charge them, I have to use a USB power adapter (charger).  It's inconvenient, but I assumed this was normal.

eried, 2024-01-29:
if that sound was not there before, maybe the charging circuitry is damaged. I would suggest removing the battery and charging it on a external charger like an "imax s6" and then checking how it behaves with a fully charged battery. Not much related with the firmware though...

u-foka, 2024-02-02:
Unfortunatelly it sounds like a hardware or battery issue :( The firmware has absolutaley nothing to do with the charging process, we cant even display the battery status on the display :/

sizzlik, 2024-02-18:
like u-foka said..its pretty sure a hardware defect in the charging circuit. If you hear a high frequency noise it might be a damaged transformer, in wich case you should not try to charge anymore and remove the battery. Serious fire hazard. If you got a multimeter you could check, but it sounds like a lost cause. When it kinda turned on while charging, making a noise and now doesnt turn on at all...just the magic smoke is missing.