portapack-mayhem/mayhem-firmware #1812: tx carrier only

CasualArclamp, 2024-01-25:
on latest 1.9.1 atm. this has been an issue in the last couple of releases. I'm not sure if it is software or my hardware. I can send pics in the future if needed. 

RX works perfectly on my porta pack H2. however when I TX it is only a carrier no matter what mode. after tx, if I go to RX there is no signal at all. does anyone know about this?
kind regards Arclamp

NotherNgineer, 2024-01-25:
Could you please list which TX apps you've tried?  Perhaps some of them work and some don't?

CasualArclamp, 2024-01-27:
> Could you please list which TX apps you've tried? Perhaps some of them work and some don't?

All tx do not work except Morse code. However all RX is fine

Brumi-2021, 2024-01-28:
Hi @CasualArclamp , 
Pls check  #1673 , or the old #466 .
And let us know about it . 
Good luck .

Brumi-2021, 2024-02-04:
Any progress news @CasualArclamp about it ? 
Did you try it , and fix it ?