portapack-mayhem/mayhem-firmware #1818: Frequency tuning issues on R9

NotherNgineer, 2024-01-26:
I notice on my HackRF r9 (latest firmware) that there are certain frequency ranges that don't seem to be tuned correctly.

If I open the Audio app, WFM mode, 200k BW, and 100k step, if I find a "birdie" on a specific frequency and increase or decrease the frequency by 100 khz, I _should_ see the birdie move left or right on the screen by 100 khz.

This works great, for example, at 2120 mhz.

However, at some frequencies, including 2159.34 mhz, it does not.  At this frequency, it seems that I have to increase or decrease the frequency by 700 khz to get the birdie to move by 100 khz.  So it seems that the wrong frequency is being tuned.  (I know there is some radio tuner code that divides by 7 that I haven't figured out yet, so it's probably related to that).

Other problematic frequencies TBD.

I do not see this issue on my older (non-r9) unit.

(May be related to issue #1772 where entering certain frequencies does seem to update the tuned frequency at all, however this is not within the frequency range mentioned in that issue.)

NotherNgineer, 2024-01-27:
Local oscillator frequencies definitely change when tuning, so I'm not sure if this is a real issue.  Closing pending further testing.

closed 2024-01-27 by NotherNgineer

jLynx, 2024-01-27:
Also need to check if you are using the TCXO or the HackRF oscillator as this can make a big difference