portapack-mayhem/mayhem-firmware #1822: WAV & IQ file converter utilities

NotherNgineer, 2024-01-29:
The Audio app and WAV Viewer apps work with 16-bit WAV files, but the Soundboard app requires 8-bit WAV files.
Perhaps a utility app that converts between different WAV files formats would be helpful.

Such an app should also support conversion between .C8 and .C16 files.

eried, 2024-01-29:
is the suggestion a new set of python converter scripts? I think the wav one is a very simple thing, but maybe a deeper solution would be to unify the WAV formats?

NotherNgineer, 2024-01-29:
> is the suggestion a new set of python converter scripts? I think the wav one is a very simple thing, but maybe a deeper solution would be to unify the WAV formats?

I was envisioning an app running on the PortaPack, not requiring a host computer.

KenVanHoeylandt, 2024-02-02:
It's probably better to have the Soundboard app able to load 16bit WAV files, just like the other apps?
That's probably what eried meant with unifying the WAV formats: the WAV code could be shared across apps. For example, to convert to 8bit on the fly.

NotherNgineer, 2024-02-06:
I've added support in PR #1849 so that the WAV & Soundboard apps can both handle 16-bit WAV files.  I'm going to consider that PR a resolution for this.

closed 2024-02-06 by gullradriel