portapack-mayhem/mayhem-firmware #1826: Tuning issue above 2740MHz?

NotherNgineer, 2024-01-29:
Above 2740MHz, it seems that one cannot transmit on one PortaPack and receive on the other.  One unit is H2R4+, and the other is H2R4+/r9.

I don't know yet if it's a transmit or receive issue, or specific to my units.  Can anyone do this successfully?

(There is a lot of noise/birdies picked up above 2740MHz, but probably all internally generated since it shows up even with the antenna removed and an SMA terminator installed.)

gullradriel, 2024-04-06:
https://github.com/portapack-mayhem/mayhem-firmware/issues/1411 is related to that issue. I closed it so we are only keeping that one as a reference.

NotherNgineer, 2024-09-24:
Unfortunately I don't think my 70MHz oscilloscope will be of much use above 2740MHz.

I have 3 PortaPacks now and just for example I tried the Microphone app on all 3 of them at F=3300MHz, NFM, gain=35, and amp=0.  It seems that I can indeed transmit from any one unit to the 2 others, but only if the receiver is within about 4 inches of the transmitter.  Previously I might have tested this when they're a foot or two apart, not realizing that the range was only a few inches.

I'll need to do more testing, but I was just wondering if others have more success.

RocketGod-git, 2025-06-11:
Tried HackRF mode -> SDRangel -> TX 
NFM & AM

Nothing received to RTL-SDR.

2400 MHz - Nothing
400 MHz - Something