portapack-mayhem/mayhem-firmware #1839: FPV drone

hackrf-user2017, 2024-02-01:
Hi all! Is it possible to make a function for scanning and viewing digital and analog video of FPV drones?;

NotherNgineer, 2024-02-01:
Anyone is welcome to try their hand at writing such an app!  Without any video processing hardware (such as MP4 decoding) it might be tough, though.  We currently don't support ATSC broadcast TV (yet) for example.  The analog TV app is the closest we currently have to viewing any sort of live video and it only supports PAL format and displays it in grey-scale 52x128 resolution.

KenVanHoeylandt, 2024-02-02:
Some ideas that might be feasible:
- Detecting drone activity (scanning certain frequency bands like [digital/analog video ones](https://learnassets.getfpv.com/learn/wp-content/uploads/2019/09/24021107/Frequency-Reference-Chart-2020-12-23-011.png) but also known frequencies for tx/rx systems.
- Showing drone telemetry (for known protocols like Crossfire, ExpressLRS, etc.)