portapack-mayhem/mayhem-firmware #1853: Speed sensitive encoder dial

NotherNgineer, 2024-02-06:
Turning the encoder dial 1000 steps takes a long time.  We could speed that up by detecting the rotation rate and increasing the number of steps per click when turned rapidly.  My old car radio supports this when searching the MP3 library.

gullradriel, 2024-02-06:
It already exists, but it is too sensitive to CPU usage.

NotherNgineer, 2024-02-07:
@gullradriel  Do you mean it was attempted at some point in the past and the code was removed?  I didn't see it in the existing encoder function hierarchy, but I haven't searched history on the relevant files yet.

gullradriel, 2024-02-07:
It was here:
https://github.com/portapack-mayhem/mayhem-firmware/pull/959
But it looks like someone or something took out 'auto' which should have value 0 from the list of the steps

NotherNgineer, 2024-02-07:
Ahhh, that code in #959 / #1860 was only for the Frequency field. I was thinking it should be implemented down in the encoder code (maybe in irq_controls.cpp) so it would apply tor ANY field, any time the encoder dial was rotated.

Other times it might be helpful include when adjusting volume, scrolling through lines in Notepad, scrolling through a list of frequencies or files, adjusting geomap zoom factor, or when adjusting the time & scale values in the WAV viewer app.

But the code in #959 / #1860 will be a helpful starting point - thank you for finding it!

closed 2024-02-10 by gullradriel