portapack-mayhem/mayhem-firmware #1855: A short time ago, I think 2 nightly versions ago the ble spam (iOS) was working but the ios crash option didn't work then, now the normal spam (iOS) isn't working, I don't think that my portapack is the problem, tried adjusting the gain, also not working, also have the latest firmware..

Bilo0344, 2024-02-07:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/portapack-mayhem/mayhem-firmware/releases)

----

**Describe the bug**
A clear and concise description of what the bug is. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'
3. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:
* Latest Nightly release:
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

Bilo0344, 2024-02-07:
It's fixed! You need to go into settings> p memory mngmt> select "use sd card for p menu> reset p menu, then it works, I looked this up in closed bugs, thanks mayham!

Bilo0344, 2024-02-07:
Sorry.. iOS crash still doesn't work.

closed 2024-02-07 by Bilo0344

htotoo, 2024-03-18:
iOS crash is fixed by Apple in the recent FWs, but it should work on older ones.