portapack-mayhem/mayhem-firmware #1868: 1.9.1 download archives do not contain external apps

yellowz1234, 2024-02-08:
**Describe the bug**
The external apps are no longer included in the download archives of Mayhem 1.9.1

**Expected behavior**
The /APPS folder should be included in the SDCARD folder or be available in a separate download.
I know that the system to deliver external apps has changed in the latest nightlies, but it seems that the original 1.9.1 download files have been overwritten and no additional OCI archive is provided for 1.9.1.
Where can we find the external apps for 1.9.1?

**Affected versions**
1.9.1

NotherNgineer, 2024-02-08:
Looks like the newly-posted 1.9.1 SDCARD zip files are missing the APPS folder...

zxkmm, 2024-02-08:
I’m sorry, I mistakenly deleted the release file this morning and when I re-upload it I forgot it.
Can someone fix it for me cuz now I don’t have any computer access? Thank you so much!

zxkmm, 2024-02-08:
I added your APPS file into release. I’ll find way to pack ZIP in iOS or can any1 have computer access do that for me?

zxkmm, 2024-02-08:
Oh I just asked @u-foka to fix it for me since he have access to computer maybe we can wait for you

zxkmm, 2024-02-08:
@u-foka just fixed it, thanks for saving my live😂

closed 2024-02-08 by NotherNgineer

yellowz1234, 2024-02-11:
Thanks all for the quick response. APPS folder is correctly included now.