portapack-mayhem/mayhem-firmware #1883: Portapack freezes

jasssoon, 2024-02-12:
Hello! I'm having an issue with my portapack freezing when I'm trying to go back to a menu of any sort. it freezes when the back arrow at the top is selected. I believe i had correctly attempted to reflash 1.9.1 and the issue still follows. I've also  attempted to  run HackRF firmware in RAM via LPC DFU and still no changes. I am out of ideas and would really like some help.

zxkmm, 2024-02-12:
Hello, would you mind try to [full reset](https://github.com/portapack-mayhem/mayhem-firmware/wiki/Full-reset) ?

jasssoon, 2024-02-12:
how would i do that?

BuddyPage, 2024-02-18:
Take out the SD card and see if it still freezes. I had a similar problem in 1.9.1 that nothing else would solve, and had to downsize from a 64gb to a 32gb SD card, and it solved it. Same problems encountered with a cheap card, so I switched to name brands. Read errors caused my problems. Hope it helps.

jLynx, 2024-02-21:
Are you still having issues on v2.0.0?