portapack-mayhem/mayhem-firmware #1885: Issue with SPFlash Mayhem Logo Location in Source Code

Petros04, 2024-02-12:
Hi Dears

 I'm reaching out regarding an issue I've encountered while trying to modify the SPFlash Mayhem program. Specifically, I'm attempting to change the program's logo, but I've run into a roadblock.

Upon inspecting the source code, I couldn't locate the SPFlash Mayhem logo file within the codebase. However, I did find it stored in the SD card folder. My objective is to change the logo within the program's driver itself, rather than relying on an external SD card.

Could you kindly provide guidance on how I can locate and modify the logo directly within the program's driver or resources?

Your assistance in resolving this matter would be greatly appreciated.

Thank you.

Best regards,Petros

NotherNgineer, 2024-02-12:
The source file containing the default splash screen bitmap is "bmp_splash.hpp"

Petros04, 2024-02-12:
Thanks for answer , if i use sd card and chang spflash icon it convert to code as a default icon or it used as a bmp picture ,and if it convert to code  can you say me where ?

Thank you !

zxkmm, 2024-02-13:
Not sure what you mean by “convert to code”.
 
If you are looking for png-> bitmap arr or bitmap arr -> png scripts, it’s under /firmware/tools.

if you are asking what would happen after you set the splash, no it won’t change any code. C++ is a compiled language, the “code” won’t be changed once compiled. If you want edit code, clone the repo and edit with an editor.

closed 2024-02-13 by Petros04